prusa3d/PrusaSlicer #7327: Adding supports sometimes cause slicer to skip some layers resulting in bad print.

MikiCZSK, 2021-11-21:
### Version
_2.3.1_

### Operating system type + version
Windows 10 20H2

### 3D printer brand / version + firmware version (if known)
This isnt problem with printer. Problem shows up in slicer.

### Behavior
Sometimes when slicing 3d model and adding manual supports, slicer will just slice support material and model part of the layer will not be sliced resulting in partialy empty layer. Should explain that. Without support generation slicer behaves as expected aka. all layers are sliced properly.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]
[image]

n8bot, 2021-11-21:
This is actually intended behaviour. The supports are being printed with varying layer height (which can be controlled in the printer settings tab).

So, some "layers" are smaller than others and contain only support. Some layers of the print contain no support. You can synchronize the layer height if there is no Z gap, or by forcing the min-max range of variable layer height in printer settings to be the exact layer height you want. Be careful, forcing the variable layer height to be one layer height can sometimes lead to missed support layers. YMMV.

rtyr, 2021-11-22:
Answered by n8bot. Closing.

closed 2021-11-22 by rtyr