prusa3d/PrusaSlicer #7328: Filename containing chars not usable on FAT result in error and crash in gcode export

qorron, 2021-11-21:
### Version
Version 2.4.0-beta1+linux-x64

### Operating system type + version
Debian Testing

### 3D printer brand / version + firmware version (if known)
MINI+

### Behavior
* have a colon ':' in the m3f file
* have a printable object on the build plate
* slice it
* insert a flashdrive with some sort of FAT on it
* hit the export to flashdrive button
* confirm the suggested filename (containing a ':')

Expected:

 * Warning regarding the problematic char
 * autoreplacement

Actual:

 * Message: "Copying of the temporary G-code to the output G-code failed. Maybe the SD card is write locked?
Error message: Invalid argument"
 * and a crash of prusa slicer upon the second time this is tired (reproducible)

_Is this a new feature request?_
not really

#### Project File (.3MF) where problem occurs
[3mf_test.zip](https://github.com/prusa3d/PrusaSlicer/files/7577066/3mf_test.zip)

Similar and yet different because of the crash to:
#4695 #3281

rtyr, 2021-11-22:
@kocikdav will look into it

bubnikv, 2021-12-08:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/7828964f8ca1c5ca029ef1ec5544769696e4eb2a

closed 2021-12-08 by bubnikv