prusa3d/PrusaSlicer #7329: Feature Request: Speed of Small Perimetes

viper0078, 2021-11-22:
### Version
2.4 beta1

### Operating system type + version
n/a

### 3D printer brand / version + firmware version (if known)
n/a

### Behavior
In some cases, the SmallPerimetes application target does not work properly. As shown in the figure, it is being applied to the blue line, which is slower than the speed of ExternalPerimetes. 

[image]

This will only prolong the printing time and is not beneficial.
In what cases is there a benefit to applying SmallPerimetes to other than ExternalPerimeter?
I think that the SmallPerimeters speed setting should only be applied to ExternalPerimeter.

_Is this a new feature request?_
yes

rtyr, 2021-11-22:
Seems like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1297.

viper0078, 2021-11-22:
No, this case is different from #1297.
This is a discussion of where to apply the SmallPerimeters speed setting.
I don't think it is necessary to apply it to the Perimeter (blue line) that exists inside.

n8bot, 2021-11-22:
Ahem https://github.com/n8bot/PrusaSlicer/commit/b419969b4ed26968e2a3b0f6d5d6530bd77ff1d1

Hehehe

viper0078, 2021-11-22:
> n8bot@b419969
Isn't that something that should be commented on in #1297?

n8bot, 2021-11-22:
Yes, but with control of the length one can enable/disable it as they please, by adjusting the length. 

Did you want greater control over it than that?

viper0078, 2021-11-22:
We want to apply it to the ExternalPerimeter, but not to the inner Perimeter, so we cannot control it by length alone.

In the modified source, the default is 40mm? Isn't that too long?
If the ExternalPerimeter is shorter than 40mm, then you want to apply it to the ExternalPerimeter, but not to the Perimeter.
This is because we believe that reducing the speed of the inner Perimeter will not contribute to improved print quality.

n8bot, 2021-11-23:
I see what you mean. Sometimes, though, like in the case of small holes, we want all perimeters to print slowly, even the internal ones.

I think your problem is with the 40 mm length, really. If you simply shorten it to be like 1/4 of the value, you'll end up with not many internal perimeters that are too slow.

One could add yet another another setting to only apply, small perimeter speed for external... but that seems a bit excessive.

The 40 mm is the default. What is the circumference of a circle of diameter 6.5 mm?

viper0078, 2021-11-23:
Even if it is a small hole, I don't think it is necessary to apply it to anything other than ExternalPerimeter.
However, I only use ABS and PLA, so I don't know about other filaments.

But that's beside the point.

It seems to me that the original purpose of this parameter is to print curves such as small circles nicely. How about a shape like the following?

[image]

If the length of the Perimeter is used as a criterion, the speed of the circle cannot be reduced as shown in the figure.
I suppose it is difficult to judge the shape of a curve, so it uses the length as a criterion, but I don't see much benefit in this.

I had a case where I wanted to slow down the printing speed partially.
This is when you draw a curve like a circle on the raft. It was difficult for the filament to settle and the shape was sometimes broken. So we thought that by slowing down the printing speed of the curved part, the fixation would be improved.
The solution is to slow down the printing speed when the printing distance between the two coordinates is short in Perimeter and ExternalPerimeter.
For example
G1 X00.000 Y00.000
G1 X00.000 Y40.000 Exx.xxxx F3600; perimeter
G1 X00.000 Y41.000 Exx.xxxxxx F300; perimeter
The idea is to use high speed for long print distances and low speed for short distances.
This is just a thought I had.

viper0078, 2021-11-23:
One more thing to add.
I usually want the SmallPerimeters option to be Off.
To turn it off, should I set the speed of SmallPerimetes to be the same as Perimeters?

If the idea is the one I wrote before, SmallPerimeters, which need to be slowed down, can be controlled by adding an AddModifiler.

n8bot, 2021-11-23:
Interesting thoughts.

First, I'm not affiliated with Prusa 3D I'm only giving you my thoughts and understanding of how it currently works, and showing how I have modified it.

I think the main purpose of the feature was to slow down when printing small islands on the first layer. Small bits of extrusion that don't have much else to hold onto.

viper0078, 2021-11-23:
I'll be honest.
I don't really see the need to apply this option except for the first layer.
However, I suppose there are people who need this option above the second layer.
I think it should be considered so that the original purpose of the option can be realized.

viper0078, 2021-11-25:
If the application is limited to the first layer, then I think slowing down all Perimeter is consistent with the original purpose.

Will this not be considered anymore and will we proceed with adding parameters?

n8bot, 2021-11-25:
I was only guessing when I said it might have been intended for the first layer. Just now I thought of another use, similar, but not on the first layer: when a small boss needs to be printed on top of some other layers. It's not the first layer, but it may be helpful for the tiny extrusions to be printed slower.

The problem with doing anything other than disabling the feature or allowing the length to be specified is that it now becomes a very involved task to change for not much benefit.

If you ask me, simply allowing for configurable length, and therefore the ability to disable the feature entirely, is the best solution.

bubnikv, 2021-12-10:
@viper0078 could you please provide your 3MF to test on?

viper0078, 2021-12-11:
@bubnikv What kind of test content do you want the 3MF file to contain?

bubnikv, 2021-12-11:
@viper0078 the one for the screenshots at the very top where the speed gets slower for inner perimeter than for the outer perimeter.

viper0078, 2021-12-11:
Attach the file.

[a.zip](https://github.com/prusa3d/PrusaSlicer/files/7696664/a.zip)

bubnikv, 2021-12-11:
I see the issue now. The problem here is that the perimeter slow down decision is made per loop, not per island. If I just trim a single island from your object, it works as expected. But in your case the external perimeter is much longer than the internal perimeters, thus the slow down logic is applied to the internal perimeters. That is not optimal.

[image]

viper0078, 2021-12-12:
I would like to see a setting to turn off the SmallPerimeterSpeed feature. setting PerimetersSpeed and SmallPerimetersSpeed to the same value does not mean it is off.
I would like to see the SmallPerimetersSpeed feature turned off, because SmallPerimetersSpeed takes precedence over BridgeSpeed.
If you set SmallPerimetersSpeed to 45mm/s and BrigeSpeed to 12mm/s, the bridge will still print at 45mm/s. SmallPerimetersSpeed is causing unintended behavior.

closed 2026-09-29 by prusaslicer-bot[bot]