prusa3d/PrusaSlicer #7331: Incorrect display of G3-commands in G-code Viewer

Overruler, 2021-11-22:
### Version
PrusaSlicer G-code Viewer
Version 2.4.0-beta1+win64

### Operating system type + version
Microsoft Windows
Version 2004 (OS Build 19035.1)

### 3D printer brand / version + firmware version (if known)
Just viewing G-code

### Behavior
The G-code Viewer does not visualize G3 commands, skipping over them. This means the resulting path is wrong.

To reproduce the issue, save the following G-code to a file and open it in PrusaSlicer G-code Viewer.
```
T0
M82
G92 E0
M109 S200
G280 S1
G0 Z20.001
G1 F1500 E-6.5
M107
M204 S625
M205 X50 Y50
G1 F600 Z2.2
G0 F1285.7 X185.622 Y141.676 Z2.2
M204 S500
M205 X20 Y20
G1 F600 Z0.2
G1 F1500 E0
G3 X179.099 Y145.546 I-6.560 J-3.625 E0.10472 F600
G1 X150.858 Y145.545 E0.47658
G3 X144.083 Y141.039 I0.032 J-7.394 E0.58997
G3 X142.471 Y136.493 I5.975 J-4.677 E0.65463
G3 X142.478 Y132.074 I90.235 J-2.067 E0.71282
G3 X142.582 Y130.846 I10.689 J0.287 E0.72906
G1 X143.425 Y124.987 E0.807
G3 X143.910 Y123.175 I8.945 J1.423 E0.83173
G3 X143.936 Y121.432 I6.685 J-0.772 E0.85474
G1 X146.464 Y103.863 E1.08846
G3 X183.974 Y104.141 I18.736 J2.708 E1.80357
G1 X187.44 Y130.973 E2.15981
G3 X187.530 Y132.437 I-12.133 J1.481 E2.17914
G1 X187.529 Y136.493 E2.23254
G3 X185.943 Y141.004 I-7.587 J-0.133 E2.29664
G1 X185.622 Y141.676 E2.30645
M204 S625
M205 X50 Y50
G0 F1285.7 X185.232 Y141.513
M204 S3000
M205 X20 Y20
M107
M82
M104 S0
M104 T1 S0

```
* _Expected Results_
The G3-commands should be displayed with the correct arcs.

* _Actual Results_
Result from a larger file showing arcs being replaced with straight lines cutting across already processed area:
[image]
Result of the code to reproduce the issue, note that there are only straight lines:
[image]

#### Project File (.3MF) where problem occurs

There is no project file, this issue occurs within G-code Viewer.

lukasmatena, 2021-11-23:
You are correct, G3 is not currently supported. It is already on the list.

bubnikv, 2021-12-08:
related to https://github.com/prusa3d/PrusaSlicer/issues/4352

PaparazziN, 2023-11-17:
Hi @Overruler!
With the release of PS-2.7.0 G2 and G3 support was added, the file you mentioned seem to look fine when I open it with the latest RC.

lukasmatena, 2023-12-03:
Correct. Implemented in 2.7.0. Closing.

closed 2023-12-03 by lukasmatena