prusa3d/PrusaSlicer #7334: Doesn't generate support with smartfill

MarkusP85, 2021-11-23:
I used smartfill to create supports but not all supports where created. 
[image]
[image]

It works with the brush but not with smartfill. 

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2021-11-23:
Please provide a 3MF and version of PrusaSlicer you have the issue with. Thanks.

MarkusP85, 2021-11-23:
[Base.zip](https://github.com/prusa3d/PrusaSlicer/files/7587127/Base.zip)
3mf isn#t supported had to zip it

> Please provide a 3MF and version of PrusaSlicer you have the issue with. Thanks.

lukasmatena, 2021-11-23:
Thanks. There was a problem in beta1 that exactly horizontal painted surfaces were sometimes ignored. It was already fixed. I reproduced your problem in 2.4.0-beta1, but not in current master, so it seems to be the same problem.
[image]
Closing.

closed 2021-11-23 by lukasmatena