prusa3d/PrusaSlicer #7337: Lilypad brim via Add Part clips the object in 2.4.0b1

stevenmhood, 2021-11-24:
### Version
2.4.0b1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
* In PS 2.3.0, when I add a part to an object to simulate a brim, but only on part of the object, it has its own perimeters.  By setting the part's perimeters to 99, it creates an easily removable brim for a corner or sharp point.
* Opening the same *.3mf file in 2.4.0b1 gives a totally different result, where the added part appears to clip the object instead of the object clipping the part.  I think this may be a result of [volume processing changes](https://github.com/prusa3d/PrusaSlicer/wiki/PrusaSlicer-2.4.0-alpha1-full-change-log#negative-volumes-modifier--extruder-override-rules).  If that's the case, is there a way to achieve the same result?
* _Expected Results_: continue to have an easily removable brim
* _Actual Results_: Brim is much more difficult to remove
  * 2.3.0 top: [image]
  * 2.3.0 bottom: [image]
  * 2.4.0b1 top: [image]
  * 2.4.0b1 bottom: [image]

_Is this a new feature request?_  It appears to be a regression (intended, according to release notes).  I'd like to propose that the `Add Part` action inserts the parts *before* the object.  I've checked a few of my projects, and this produces the expected behavior.  However, I realize there may be other use cases that I'm not considering.  If those exist, I'd like to learn about them and figure out how we can make this behavior more intuitive or obvious to the user.

#### Project File (.3MF) where problem occurs
[Makita DC18RD Charger Bracket - brim-demo.zip](https://github.com/prusa3d/PrusaSlicer/files/7592471/Makita.DC18RD.Charger.Bracket.-.brim-demo.zip)

lukasmatena, 2021-11-24:
> I'd like to propose that the Add Part action inserts the parts before the object

You can reorder the parts yourself using drag-and-drop in the object list.

stevenmhood, 2021-11-24:
> You can reorder the parts yourself using drag-and-drop in the object list.

Yes, I *can*, but my question is whether adding parts ahead of the object will preserve what users are doing *and* be compatible with the new volume processing in 2.4.  Reordering isn't something I've ever had to do in over a year of using PS.  I didn't even know it was possible until I read the release notes *very* carefully to realize that ordering mattered, thus the suggestion.  How do we make this intuitive for users?

bubnikv, 2021-11-30:
> How do we make this intuitive for users?

Through a documentation.

A usual convention of CAD systems is that the latter object overrides the former one. This is the rule PrusaSlicer 2.4 follows, whihc unfortunately breaks backward compatibility. Also for multi-material prints PrusaSlicer 2.3.3 was not very predictable while PrusaSlicer 2.4.0 follows a very predictable single rule when deciding what overrides what.

bubnikv, 2021-12-08:
Answered, closing.

closed 2021-12-08 by bubnikv