prusa3d/PrusaSlicer #7339: bulge on exterior walls when there are top layers on the other side of the wall

ssill2, 2021-11-24:
### Version
_Version of PrusaSlicer used goes here_
2.3.3 and I believe even in the latest 2.4.0 builds.   I built from master yesterday and I just sliced the same model with 2.4.x to see if the issue persists in latest.   I've got the print going on two different prusa printers so I'll report back with screenshots.

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
windows 10, windows 11 and others

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa MK3s+

### Behavior
* _Describe the problem_
In simple models where there is a top surface surrounded by walls, the resulting print has a bulge on the outer wall  directly opposite the top surface.  This issue is documented heavily in the following prusa forum thread
https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-how-do-i-print-this-printing-help/buldge-when-print-reaches-solid-layers/paged/1/
Numerous screenshots there illustrate the issue.

_Is this a new feature request?_
No

Here's a 3mf that exhibits the issue
[makeup-pot-with-lid.zip](https://github.com/prusa3d/PrusaSlicer/files/7596637/makeup-pot-with-lid.zip)

[image]

ssill2, 2021-11-24:
I'm not sure why my image didn't come through.  This has happened on the prusa forums recently since I upgraded to win 11.

ssill2, 2021-11-24:
Here's one I took using snip and sketch tool instead of the one off my iPhone.  
[image]

ssill2, 2021-11-24:
so I can confirm that even when sliced with a freshly built from master 2.4.0,  the bulge is still present during print even though the preview in the slicer doesn't show the bulge.
[image]
you can see the bulge is directly across from the top layers in the middle of the model.

rtyr, 2021-11-25:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/991, https://github.com/prusa3d/PrusaSlicer/issues/1719, https://github.com/prusa3d/PrusaSlicer/issues/3423, https://github.com/prusa3d/PrusaSlicer/issues/4010, etc.

See this comment for further information https://github.com/prusa3d/PrusaSlicer/issues/991#issuecomment-458614931. It is very difficult to compensate for all the influences mentioned there, but improvements related to this topic are on our todo list.

ssill2, 2021-11-25:
I'll close this as a dupe then, thanks!

closed 2021-11-25 by ssill2

ssill2, 2022-11-11:
> Hi, any update on this please? I'm still having this issue with
> PrusaSlicer 2.5.0

Same unfortunately

presling, 2024-07-22:
Here's a YouTube video explaining how to fix that issue. 
https://youtu.be/8vpsqQx0iig?si=lvoKOTe5lp3bkvXB