prusa3d/PrusaSlicer #7340: Support on top of brim when wipe tower enabled

rajapinta, 2021-11-24:
### Version
Version 2.4.0-beta1+win64
PrusaSlicer-2.4.0-beta1+win64-202110251333

### Operating system type + version
Microsoft Windows 10 Pro
Version 10.0.19043

### 3D printer brand / version + firmware version (if known)
Custom FDM Multi Extruder

### Behavior
Brim calculation changes when wipe tower activated. Could we have support on top of brim when wipe tower activated.

When wipe tower and support activated for dual extruder. Extruder 1 for build material, and extruder 2 for support material. 

Brim is correctly shown in Feature view

[image]

But it seems that brim is printed incorrectly with extruder 2.

[image]

When wipe tower disabled, brim is printed correctly with extruder 1.

[image]
[image]

Related request
#713 #1156 #3395 #3396.

bubnikv, 2021-12-08:
> Could we have support on top of brim when wipe tower activated.

I don't understand what you mean.

Do you suggest that the brim is printed with support material? Or the object material? With the support material always?

rajapinta, 2021-12-08:
> I don't understand what you mean.
> 
> Do you suggest that the brim is printed with support material? Or the object material? With the support material always?

Yes. When wipe tower is activated then support material is used to print brim. At least for me when having two extruders one for support and another for object material.

bubnikv, 2021-12-08:
> Yes. When wipe tower is activated then support material is used to print brim. At least for me when having two extruders one for support and another for object material.

So you expect the brim to be printed with the same material as the object in every case?

rajapinta, 2021-12-08:
> So you expect the brim to be printed with the same material as the object in every case?

Yes, that is my expectation.

bubnikv, 2021-12-08:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/6979 then

closed 2026-09-29 by prusaslicer-bot[bot]