prusa3d/PrusaSlicer #7341: BUG: Unable to build with CGAL 5.X

theofficialgman, 2021-11-25:
### Version
Building on current master.
https://github.com/prusa3d/PrusaSlicer/blob/4c9ea5241dd2e1856f3e6f573b35ca46bafa9780/src/libslic3r/CMakeLists.txt#L291
https://github.com/prusa3d/PrusaSlicer/commit/83fb066c645ce25db1720314fd82ed0e62dfa57f

The above recent commit locks CGAL to only 4.13.X versions while most OSs packages use 5.X (see an explaination at stackoverflow https://stackoverflow.com/questions/7855174/set-minimum-version-of-boost-in-cmake)
basically specifying the minor version number locks in 4.X.X versions only, no 5.X.X will satisfy the command (even though they do work)

tamasmeszaros, 2021-11-25:
Thanks you @theofficialgman its my bad.

theofficialgman, 2021-11-26:
@tamasmeszaros this still will not work based on the CGALConfigVersion.cmake file (example below)

This still requires the major version (4) to match as well as have a minor version equal or greater than 13
(so 4.13, and 4.14 will work but 5.0, 5.1, etc do not work)

I'm not sure exactly which versions of cgal prusaslicer will actually function with, but it at least compiles for me on cgal 5.0 IF I remove the check
I would suggest using the version rage format for find_package but that was added in cmake 3.19, and prusaslicer currently only requires 3.13 to build
```
A version range with the format versionMin...[<]versionMax where versionMin and versionMax have the same format and constraints on components being integers as the single version. By default, both end points are included. By specifying <, the upper end point will be excluded. Version ranges are only supported with CMake 3.19 or later.
```
```

set(CGAL_MAJOR_VERSION 5)
set(CGAL_MINOR_VERSION 0)
set(CGAL_BUGFIX_VERSION 4)
set(CGAL_VERSION_PUBLIC_RELEASE_VERSION "5.0.4")
set(CGAL_VERSION_PUBLIC_RELEASE_NAME "CGAL-${CGAL_VERSION_PUBLIC_RELEASE_VERSION}")

if (CGAL_BUGFIX_VERSION AND CGAL_BUGFIX_VERSION GREATER 0)
  set(CGAL_CREATED_VERSION_NUM "${CGAL_MAJOR_VERSION}.${CGAL_MINOR_VERSION}.${CGAL_BUGFIX_VERSION}")
else()
  set(CGAL_CREATED_VERSION_NUM "${CGAL_MAJOR_VERSION}.${CGAL_MINOR_VERSION}")
endif()

set(PACKAGE_VERSION ${CGAL_CREATED_VERSION_NUM})

if(PACKAGE_VERSION VERSION_LESS PACKAGE_FIND_VERSION)
  set(PACKAGE_VERSION_COMPATIBLE FALSE)
else()
  if(PACKAGE_FIND_VERSION_MAJOR STREQUAL CGAL_MAJOR_VERSION)
    set(PACKAGE_VERSION_COMPATIBLE TRUE)
  else()
    set(PACKAGE_VERSION_COMPATIBLE FALSE)
  endif()
  if(PACKAGE_FIND_VERSION STREQUAL PACKAGE_VERSION)
    set(PACKAGE_VERSION_EXACT TRUE)
  endif()
endif()
```

theofficialgman, 2021-11-26:
honestly I feel like it would just be better to go back to the old way and have compilation fail if the libraries are missing at this point

theofficialgman, 2021-11-26:
@tamasmeszaros actually I think this should work for both:
```
find_package(CGAL 4.13)

if ( NOT ${CGAL_FOUND} )

    find_package(CGAL 5.0 REQUIRED)

endif ()
```
it will still fail as required and allow both major versions
the first one is non-fatal to allow the second check to run
versions under 4.13 will fatal error on the second check

clytle374, 2021-11-28:
This issue has my build broken.

closed 2021-12-03 by tamasmeszaros

tamasmeszaros, 2021-12-03:
> honestly I feel like it would just be better to go back to the old way and have compilation fail if the libraries are missing at this point

@theofficialgman I aggree

commit 2f805de4 (2021-11-25): Fix locked in version of CGAL

fixes #7341
commit 08e3e60a (2021-12-03): Fix CGAL build with major version > 4 on Linux 

fixes #7341

CGAL upstream CMake config will lock in the major version if installed as a distro package and will not provide any version info if compiled and installed from upstream. As of this commit, PrusaSlicer can be built with CGAL 4.13.2, and 5.0 but there is no universal way to specify the minimum version that would work with static dependencies and linux packages.

sorry #4912