prusa3d/PrusaSlicer #7344: Feature Request: Ribbed vault supports

photodude, 2021-11-25:
### Version
Version 2.3.3+win64

### Operating system type + version
OS Name	Microsoft Windows 10 Home
Version	10.0.19044 Build 19044

### 3D printer brand / version + firmware version (if known)
BIBO touch 2

### Behavior
_Is this a new feature request?_

Just saw the Concept of "ribbed vault supports" aka "lightning supports" This seems like a great feature to add. 

Link to the source paper which includes code
https://hal.archives-ouvertes.fr/hal-02155929/document

Link to a video discussing the paper and its use in other slicers
https://youtu.be/BAtqjN2l65M

dartrax, 2021-11-26:
@rtyr looking at the pictures it‘s more about infill than (outer) support structure.

rtyr, 2021-11-26:
You are right, I was too quick with the labels.

photodude, 2021-11-27:
@dartrax Yes, infill supports. I've added some additional text to clarify.

jaggzh, 2021-12-29:
Yeah, these are precisely what I end up designing by hand, both internal in objects, as well as externally.  It's a pain to make them by hand.  For supports, instead of infill, they'd only need some other otions to let them be removable, like:

Ribbed/fin supports ->
1. -> Bottom joint height (maximum height of fins to bond to the object wall)
2. -> Bottom joint distance (distance from wall, default .1mm, to allow fins to bond but be removable)
3. -> Rib non-bonding distance (distance from wall-to-rib for the non-joining parts)
4. -> Rib-top distance (Two options: Defaults to the Support Z and XY distance, or we can just use those, and not have two options here)
5. ... not sure what else would be good here.

jaggzh, 2021-12-29:
https://www.fabbaloo.com/news/ultimaker-curas-secret-lightning-infill-really-works

jaggzh, 2021-12-30:
https://github.com/Ultimaker/Cura/issues/11151
https://github.com/Ultimaker/Cura/issues/11155

bubnikv, 2022-06-29:
Ported from Cura to PrusaSlicer 2.5.0-alpha2. Thanks Cura team for keeping Cura open source and for their effort and dedication.

closed 2022-06-29 by bubnikv