prusa3d/PrusaSlicer #7345: Incorrect preview

kb-, 2021-11-25:
### Version
Version 2.3.3+win64

### Operating system type + version
PrusaSlicer
Version:   2.3.3+win64
Build:     PrusaSlicer-2.3.3+win64-202107161027

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 8.1 (build 9600), 64-bit edition
Total RAM size [MB]: 8,473MB
OpenGL installation
GL version:   4.6.0 NVIDIA 425.31
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 660M/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
[... 274 lines cut ...]
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### Behavior
Some strange aliasing effect occurs, depending on zoom level. It makes more or less big lines non parralel to print bed.

[image]

[image]

rtyr, 2021-11-26:
Unfortunately, we cannot control such moire effects.

For more information, please see:
https://en.wikipedia.org/wiki/Moir%C3%A9_pattern

Closing.

closed 2021-11-26 by rtyr

n8bot, 2021-11-29:
I have good news. There are multiple factors at play here, and with the right graphics card (Nvidia, probably AMD too) you can control these settings to get much nicer look.

## First, most importantly, is to disable MFAA. With a Nvidia card, you do this by overriding global settings on a per-application basis.
[image]
[image]

## This will get rid of the bright white/oversaturation at far away zoom distances. Still, the moire will be bad. So, for that, add as much *normal* anti-aliasing as you can. Avoid FXAA and especially avoid the MFAA that we just disabled.
[image]
[image]
[image]

I think think this is is referred to as multisample (or multisampling) in the codebase. It might provide some benefit for some systems, I'm not sure, but it causes massively weird "overexposure" at extreme zoom settings which contributes to the ugly moire effects.

n8bot, 2021-11-29:
I got it looking really good
[image]

bubnikv, 2021-11-29:
@enricoturri1966 what do you think?

bubnikv, 2021-11-29:
The original issue clearly showed no anti aliasing provided by the graphics system at all, thus it must have been a very old system.

n8bot, 2021-11-29:
> The original issue clearly showed no anti aliasing provided by the graphics system at all, thus it must have been a very old system.

I had to enable AA forcefully in Nvidia panel to get it. I never even thought about it before.

I'm no expert, but it seems that "FXAA" and "MFAA" are both more detrimental to the scene than they are helpful.

Also check out my latest release omg so many colours.

bubnikv, 2021-11-29:
PrusaSlicer asks OpenGL for MSAA. It should be enabled if supported by the
system.

n8bot, 2021-11-29:
Ah, yes, that is what I thought but was not sure. I thought maybe it was for retina displays.

Nvidia themselves say that FXAA can cause artifacting near edges of shapes. Maybe opengl is selecting FXAA? Is FXAA and MFAA combination possible? That seems like it would be the "problem" in the sense that it doesn't work very well at preventing aliasing.

I notice a much better improvement in the moiré with the standard AA.

enricoturri1966, 2021-11-29:
I made a few tests.
As can be seen from the following screenshots, I found no impact on the visual appearance when changing AA settings.

[image]
[image]
[image]
[image]

n8bot, 2021-11-29:
Yes, of course it won't look better by ONLY disabling MFAA. However, MFAA needs to be disabled in order for weird lighting artifact from shining around when you are zoomed all the way out and looking at the right angle to a surface to illicit shinniness.

n8bot, 2021-11-29:
### Stock behaviour, no gpu ovverrides:

Missing lines in bed texture:
[image]

Visible yellow/off-colour artifacting around edges
[image]
[image]

Weird specular artifacting:
[image]
[image]

### Disabling MFAA only:

Lines no longer missing, but obviously no anti-aliasing happening:
[image]

No more yellow edge artifact (but, still, lots of aliasing):
[image]
[image]
[image]

However, **no weird specular artifacts** (but lots of normal aliasing artifacts):
[image]
[image]
[image]
[image]

**So now, the weird part:** If we now disable MFAA, while enabling normal AA, we solve the specular artifacts in the slice preview, as well as the missing line on bed texture, but we still encounter the yellow artifact on the edges. Also, the anti-aliasing in general is sub-optimal.

The magic setting appears to be this `Antialiasing - Transparency` setting. Making sure it is off MFAA, and on max, ensures the nicest image quality in my experience.
[image]

### Once I set that transparency setting to max supersampling (the importance is probably disabling the multisampling), and put everything else to best quality settings like anisotropic filtering, etc, while keeping FXAA and MFAA disabled, the best image quality is achieved. Dramatically improved:

No specular artifacts, no edge colour artifacts, beautiful smooth edges, and most importantly minimal moiré
[image]
[image]
[image]
[image]

n8bot, 2021-11-29:
The final settings I used to get the best results:
[image]
[image]
[image]

n8bot, 2021-11-29:
Sorry to flood with messages... last thing, sorry. I had the thought that maybe the reason the transparency antialiasing setting is important is because essentially the whole viewport is rendered looking through a transparent texture, no? The imgui interface? Or just the mere presence of transparent texture elements triggers some kind of artifact.

Whatever it is... it can be greatly reduced with some cards/settings.

n8bot, 2021-11-29:
> I made a few tests. As can be seen from the following screenshots, I found no impact on the visual appearance when changing AA settings.
> 
> [image] [image] [image] [image]

Oh, by the way, it looks like you were not restarting prusaslicer in between changing settings? I found that this is necessary for the new settings to take effect. Also, when disabling an override profile, simply deleting it does not work. You have to disable settings explicitly before deleting profile. FYI.

n8bot, 2021-11-29:
After getting some sleep and some coffee, and re-trying all the combinations, I think I figured out what is good and not good in the Nvidia settings panel:

- We want higher AA. The higher the AA, the less moiré.
- We *definitely* want to enable that `Antialiasing - Transparency`. I have only experimented with the maximum 8x supersampling -- perhaps the other lower options provide benefit too. This settings allows the bed grid texture to be flawless, and also alleviates the moirê the most... I think? It's still hard to determine exactly what is going on.
- `FXAA` we want off. Always. It's always worse when on.
- `MFAA` seems to have not much impact. The reason switching this setting alleviates the specular artifacting is likely due to some other configuration change that this setting induces. This was a red herring.

Tl;DR: Antialiasing GOOD, `Antialiasing - Transparency` BETTER.

However, it seems that without setting maximum antialiasing in general, the transparency setting is not much use. This definitely seems like a job for per-user settings in their gpu configuration panel, and the current prusaslicer behaviour seems acceptable. I'm not sure how much it would be worth trying to get the different gpus to specify specific antialiasing settings as default. It seems like when opengl is told to enable multisample, it picks whatever settings combination it wants (doesn't seem to be the default global settings?)