prusa3d/PrusaSlicer #7351: How hard would it be to convince you that float is not sufficient for GCodeProvessor::AxisCoords, ?

n8bot, 2021-11-26:
Or, instead, we could just:
```cpp
        if (m_wiping)
            type = EMoveType::Wipe;
        else if (!line.has_e() && (line.has_x() || line.has_y() || line.has_z()))
            type = EMoveType::Travel;
        else if (line.has_e() && (line.has_x() || line.has_y() || line.has_z()))
            type = EMoveType::Extrude;
        else if (line.has_e() && delta_pos[E] < 0.0f)
            type = EMoveType::Retract;
        else if (line.has_e() && delta_pos[E] > 0.0f)
            type = EMoveType::Unretract;

        return type;
```

Because otherwise the G-Code preview is absolutely messed up with tiny E, or X/Y/Z deltas. Absolute E position is insufficiently tracked, and therefore the deltas get progressively worse as the print goes on. Smaller layer heights/extrusion width combos exacerbate this issue, because the precision of the float gets taken up by the larger digits (left of decimal). So the delta becomes completely errored out eventually.

rtyr, 2021-11-26:
> Because otherwise the G-Code preview is absolutely messed up with tiny E, or X/Y/Z deltas.

Could you please provide some example 3MF?

n8bot, 2021-11-26:
On stock PrusaSlicer 2.4.0-beta1:
[image]
[image]
Missing perimeters completely:
[image]

I will admit that this issue is more prevalent in my branch because I have decreased the douglas_peuker threshold for simplification of perimeters to make them best quality possible. However, at times, it does pop its head up in Stock Prusaslicer. As layer height gets small, with small nozzles, and the mesh resolution is very fine, it happens.

Here is the 3mf
[LargeDetailedSphere.zip](https://github.com/prusa3d/PrusaSlicer/files/7610458/LargeDetailedSphere.zip)
:

n8bot, 2021-11-26:
With my patch on my branch (so, extra precision too. Without my patch, on my branch, the top layers would be nearly invisible in the preview):
[image]

rtyr, 2021-11-26:
You are using relative extrusion distances, therefore you should have G92 E0 command in "before layer change g-code" section.  

The firmware internally maintains an absolute counter even if it is addressed relatively in the g-code. The same applies to firmware simulator in the PrusaSlicer. This G92 E0 resets the internal counter, thus preventing the accumulation of errors.

All you need to do is to fix your printer profile (add G92 E0 command). We are considering emitting this command automatically with relative extrusion enabled.

See:
https://github.com/prusa3d/PrusaSlicer/issues/4881#issuecomment-709853417

Similar issues:
https://github.com/prusa3d/PrusaSlicer/issues/6373
https://github.com/prusa3d/PrusaSlicer/issues/6336
https://github.com/prusa3d/PrusaSlicer/issues/5073

n8bot, 2021-11-26:
That is not the issue. The issue is exactly the internal tracking of the absolute counter.

The prints are all perfect. No G92 is needed -- this is relative extrusion after all. The problem lies solely in the visualization.

n8bot, 2021-11-26:
Ok, so, you are right in so many ways.

I mean, the classification of moves works either way, but what works *better* is to reset the E position on every layer change, regardless of anything else:

https://github.com/n8bot/PrusaSlicer/commit/facd686a73f5dcb0e1a79898e60a7ba370faeb24

Because this fixes the missing corner caps that I was still seeing! And, I don't *need* to change the classification, so I won't.

Thanks for showing me the way.

But, please, don't force G92 E0 into the g-code. It's not needed except for absolute extrusion mode.

bubnikv, 2021-11-29:
You should dig into Marlin firmware source code as well to understand why G92 E0 at each layer is necessary with relative extruder addressing mode. PrusaSlicer G-code viewer relies on that as well, however it just matches the accumulation issues that happen in the firmware. With G92 E0 both PrusaSlicer and Marlin firmwares are happy.

n8bot, 2021-11-29:
Yes, I do remember hearing long ago with problems with ancient marlin firmware. It can't still be a problem today, can it?

If the firmware is fundamentally flawed, why are we insisting on matching that behaviour in the slicer?

RepRapFirmware, as an example, even when using *relative* extrusion, internally uses an absolute counter to keep track of the extruder position. Obviously, they use a datatype of sufficient size to track the E position, because no G92 E0 is needed to ensure accurate E calculation.

As for Marlin firmware... I can not imagine that the current versions are susceptible to such an easy-to-solve error. I might go digging around to see if they use a float instead of a double for tracking the absolute E position... but I doubt they use float! Maybe in marlin from 2015.

bubnikv, 2021-12-02:
> RepRapFirmware, as an example, even when using relative extrusion, internally uses an absolute counter to keep track of the extruder position. Obviously, they use a datatype of sufficient size to track the E position

Marlin does the same. What is the data type used by RepRapFirmware? Marlin uses 32bit float.

n8bot, 2021-12-02:
Lol... well, as it turns out the variable in RRF is *only* used to track the absolute extruder position for the calculation of total filament print statistics. (I guess it uses the literal values from G-code in calculating moves).

So, yes, this variable is also 32 bit float in RRF -- but it does not impact the printing.

Does the 32 bit float that accumulates absolute E position, in marlin, impact the printing still?

bubnikv, 2022-02-04:
Fixed with 1fb41a8e2242064a893ec1d03b3c3181e7232fa1
    #4881 - #5073 - #6336 - GCodeProcessor::AxisCoords using doubles in place of floats

closed 2022-02-04 by bubnikv

n8bot, 2022-02-04:
Thank you. The amount of confusion on my part was cringe-worthy, looking back.

At least, now, I kind of know where to look in Marlin so I can be more informed when discussing issues. Even my understanding of RRF was cringe-worthy two months ago.