prusa3d/PrusaSlicer #7353: Missing retract on support perimeter on 1st layer and unnecessary support field on 1st layer

msteini82, 2021-11-26:
### Version
2.4.0-beta1

### Operating system type + version
macOS Big Sur Version 11.6 (Build 20G165)

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S+ (MK3_3.10.0)

### Behavior
When printing the new added single perimeter around support on the 1st layer (added in https://github.com/prusa3d/PrusaSlicer/commit/fcb714cd241f0847522b9aac57da6b62acdd0152) there is no retract if there is a hole in the support and the nozzle changes its position. In the added project file should be a retract before the gcode step 291 on the first layer. Without retract the inner perimeter is not printed clean.

And there is a not necessary support field on the first layer in my project, which hasn't any support on the following layers. That shouldn't be generated  ;-) 

[image]

#### Project File (.3MF) where problem occurs

[test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7610505/test.3mf.zip)

neophyl, 2021-11-26:
The extra support base looks like its caused by the First Layer expansion value of 3mm that is set.  It only happens on one side as your oval is not symmetrical left/right (as oriented on your build plate).  Set a first layer expansion of 0 and it goes away.

The missing retract might be needed but from looking at the preview the nozzle doesnt seem to cross any area where the object is printed, only that support base layer.  Which if a string of filament is laid down there from a leaky nozzle then at least its only on the support.  It does appear to retract when moving from each island of support to the next, just not when printing within each support island.

n8bot, 2021-11-26:
This is the coded functionality: suppress retraction when travelling within support islands:

https://github.com/prusa3d/PrusaSlicer/blob/2c5ddfefb2e78938766d6d02c01000f0d99ea49f/src/libslic3r/GCode.cpp#L3079
```cpp
    if (role == erSupportMaterial) {
        const SupportLayer* support_layer = dynamic_cast<const SupportLayer*>(m_layer);
        //FIXME support_layer->support_islands.contains should use some search structure!
        if (support_layer != NULL && support_layer->support_islands.contains(travel))
            // skip retraction if this is a travel move inside a support material island
            //FIXME not retracting over a long path may cause oozing, which in turn may result in missing material
            // at the end of the extrusion path!
            return false;
    }
```

msteini82, 2021-11-27:
It's right, the extra support base goes away if I increase the expansion value, the model was just a quick test model for showing the missing retract.

The nozzle is clean and the problem is exactly what is written in the FIXME comment on line 3084 in the code part, there is missing material on the inner perimeter:

[image]

But I'm glad to see that there is already an FIXME comment in the code 😊

bubnikv, 2021-12-10:
The leakage of the first support is also fixed with the snug supports

[image]

and decreasing the grid pattern spacing to 1.5

[image]

gyscos, 2023-01-25:
I can see a lack of retraction when the head needs to move from one support to another, in my case across the entire build plate. This (as expected) results in large globs/filaments left on the print itself, as well as under-extrusion at the start of the support.

athompson673, 2024-01-28:
I am having this same issue, and am disappointed to see it has been known about for a long time but not fixed. In particular, I have an island with several holes in it where the part itself touches the build plate. These form outer loops for the support around them, which are all part of the same island. Without z hop between them, travels between these loops cause some to get knocked off by the small amount of oozing during the travel.

[image]

I was able to mostly solve this by re-orienting the part so it alternates between islands when printing the loops which causes a retraction, but I would love at least some sort of config option to skip the following check: https://github.com/prusa3d/PrusaSlicer/blob/97c3679a37e9ede812432e25a096e4906110d441/src/libslic3r/GCode.cpp#L3536-L3555

closed 2026-09-29 by prusaslicer-bot[bot]