prusa3d/PrusaSlicer #7355: PrusaSlicer 2.4.0-beta-2+win64 will not start on Windows 7 64-bit

jcwren, 2021-11-27:
### Version
2.4.0-beta-2+win64

### Operating system type + version
WIndows 7 Professional SP1 64-bit

### Behavior
PrusaSlicer fails to start, throws error GetDpiForWindow not found in USER32.DLL

[image]

vlan-sourse, 2021-11-27:
I have the same problem 
Version
2.4.0-beta-2+win64

Operating system type + version
WIndows 7 Professional SP1 64-bit

lukasmatena, 2021-11-28:
Thanks. Apparently it does not work with Windows 7. We'll fix it.

RUMIKUB, 2021-11-28:
sur win32 cela ne fonctionne pas non plus     comment être prévenu lors de la réparation

Psychograp, 2021-11-29:
Hi, similar issue on windows 8.1 x64
[image]

lukasmatena, 2021-11-29:
@Psychograp Yes, that is not surprising. According to [Microsoft Docs](https://docs.microsoft.com/en-us/windows/win32/api/winuser/nf-winuser-getdpiforwindow), the function requires at least
> Windows 10, version 1607 [desktop apps only]

Anything older will be affected. @YuSanka is already working on a fix.

Hardwarekiller, 2021-11-29:
I Have the Same Problem on Win7 Pro
GetDpiForWindows User32.dll 

Thanks

Hardwarekiller

bubnikv, 2021-12-04:
Fixed with PrusaSlicer 2.4.0-beta3.

closed 2021-12-04 by bubnikv

Hardwarekiller, 2021-12-04:
Thanks, now its working 

Hardwarekiller