prusa3d/PrusaSlicer #7356: Sliced G-Code miscalculates a layer to be under the bed due to custom After-Layer G-Code

cloud30000, 2021-11-27:
### Version
Version 2.4.0-beta2+win64

### Operating system type + version
Wiindows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+

### Behavior
Bug:
When specific custom After-Layer G-Code exists which is resolved to "false" in a "if" boolean and therefore not used, the sliced G-Code includes a G1 command after the end of the After-Layer G-Code to move the nozzle to a position under the bed, which corresponds to some of the unsed After-Layer G-Code.

The Slicer appears to still interpret the Nozzle Position based on the unused code and somehow think the nozzle should be or already is under the bed, and must be moved back up to the bed during the extrusion of the next 20 commands/200mm or so.

The specific code in the After-Layer G-Code (abridged) is:

```
{if ((layer_num==1) || ((complete_objects)&&((first_layer_height<layer_z) && (layer_z<(first_layer_height+(layer_height*2))))))}M83 ; extruder relative mode
G1 E-5 F{retract_speed[0]*60}
G91 ; relative positioning
G0 Z20 F{travel_speed_z*60}
;...do stuff
G0 Z-20 F{travel_speed_z*60}
G90 ; absolute positioning
G1 E5 F{retract_speed[0]*60}
{endif}
```

```
One workaround to avoid the bug is to replace the relative Z moves with absolute moves instead:
{if ((layer_num==1) || ((complete_objects)&&((first_layer_height<layer_z) && (layer_z<(first_layer_height+(layer_height*2))))))}M83 ; extruder relative mode
G1 E-5 F{retract_speed[0]*60}
G0 Z{z_offset+first_layer_height+20} F{travel_speed_z*60}
;...do stuff
G0 Z{z_offset+first_layer_height} F{travel_speed_z*60}
G1 E5 F{retract_speed[0]*60}
{endif}
```

#### Project File (.3MF) where problem occurs
[Layer starting below bed.zip](https://github.com/prusa3d/PrusaSlicer/files/7611194/Layer.starting.below.bed.zip)
[image]

lukasmatena, 2021-11-28:
Reproduced, interesting. This is somehow connected to spiral vase. It is already present in 2.3.1.

bubnikv, 2021-12-10:
The spiral vase works by processing a layer with a G-code post processor. The spiral vase G-code post-processor expects all the moves to be addressed in absolute mode, while the extruder should be addressed with relative mode.

We may improve one day, but this is really an edge case.

closed 2026-09-29 by prusaslicer-bot[bot]