prusa3d/PrusaSlicer #7358: Unhandled unknow exception; terminating the appplication

zbkingCZ, 2021-11-27:
### Version
Version 2.4.0-beta2+win64

### Operating system type + version
Windows 10 21H2 build 19044.1348

### 3D printer brand / version + firmware version (if known)
Prusa I3 MK3S+ fw 3.10.0

### Behavior
0. Start Prusa Slicer 2.4.0-beta2 (IMPORTANT!!! Czech language setting)
1. Import specific model (example in attachment)
2. Place surface, not necessary
3. "Slice now"
4. Error message: "Unhandled unknow exception; terminating the appplication"  (Picture_01) / OK
5. Error message: "Vnitřní chyba: boost::_many_args: format-string referred to fewer arguments than were" (Picture_02) / OK
6. The program closes itself.

Then I tried to switch slicer language settings to English and everything was OK. Model is sliced perfectly. (Picture_03)

#### Project File (.3MF) where problem occurs
The model was created in OnShape. The problem is only with the cap. 
https://cad.onshape.com/documents/d97177fb8529f28312c8845a/w/235e627fd8a6ba99cc868282/e/b8de7142d0f4ae1d48c6582c

Attachments in Zip file:
Picture_01.png
Picture_02.png
Picture_03.png
Toilet Freshener-Cap V03.stl
Toilet Freshener-Cap V03_Czech_lang.3mf    - Czech Language Settings
Toilet Freshener-Cap V03_English_lang.3mf    - English Language Settings

#### Thanks to
By the way, I have to commend you for your good work. I'm already looking forward to the new version of Prusa Slicer. Really a lot! :)

[Prusa Slicer 2.4.0 Beta 2.zip](https://github.com/prusa3d/PrusaSlicer/files/7611745/Prusa.Slicer.2.4.0.Beta.2.zip)

lukasmatena, 2021-11-28:
Thanks. This is a bug in Czech translation, not in the application itself. It will be fixed in the next release.

bubnikv, 2021-12-08:
Fixed in beta3. Closing

closed 2021-12-08 by bubnikv