prusa3d/PrusaSlicer #7359: SLA Supports - Manual editing ISSUE

angelozap, 2021-11-27:
### Version
2.4.0-beta2+win64

Build
202111261

### Operating system type + version
Windows 11

### Behavior
*_Is impossible to place manual supports when using the SL1 slicing mode. When you select "SLA Support Points" and then "Manual Editing" if you try to place **points** under the model or on the external/internal suface no point are placed. Furthermore, in this example I was trying to place supports on a funnel model and if clicking inside the object points are added "according to the view! but not on the surface. Infact when rotating the view the points are floating in the 3D space. After supports are generated they are touching nothing. The issue occurs on all the models I'm trying to slice. _

* _Add a model while in SLA mode; position it optimizing the orientation with "Reduced overhang slopes"; try to apply supports by clicking on Manual editing (it occurs even if Auto-generated supports are made)_
 
* _Manual supports "wherever" I decide to place them_
* _No support points, or floating support heads are generated_
  

#### Project File (.3MF) where problem occur
[funnel_support_points_issue_.zip](https://github.com/prusa3d/PrusaSlicer/files/7611749/funnel_support_points_issue_.zip)
[image]
[image]
[image]
[image]

lukasmatena, 2021-11-28:
Thanks. This looks similar to #6100 and #6744. This one is even better in that when you click "reset rotation" button while editing, PrusaSlicer crashes.

We'll fix it, but not in 2.4.0 release. I just checked and it is already present in 2.3.1. We will look at the cause of the crash, though.

stepikovo, 2022-03-14:
I'm closing it as a duplicate of #6100
This issue should be fixed in 2.5.0

closed 2022-03-14 by stepikovo