prusa3d/PrusaSlicer #7360: Prusaslicer fails to slice a simple box object S3D slices it properly

old-relic, 2021-11-27:
### Version
Ver 2.3.3

Version 2.3.3+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows 10 ver 21H1
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
CR10-S4 Duet2-WiFi

### Behavior
* Prusaslicer produces a truncated gcode file, the object is a simple box open at the top. The box should be about 400+ layers tall at .3mm layer height. Prusaslicer outputs a file that generates a box with walls about 125 layers high. The same STL file processed by S3D generates a gcode file that builds the box full height.
* 
* _Steps needed to reproduce the problem_
 Slice the attached file with .5mm nozzle at 3mm layer height. Send output to printer via wifi Preview looks correct gcode produced is too short. 
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_ no it's a bug report

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)
[image]
[image]
_
[A123 battery box.zip](https://github.com/prusa3d/PrusaSlicer/files/7611946/A123.battery.box.zip)

fe60, 2021-11-28:
I can't reproduce the issue with the shared stl-file.
Can you provide the whole project you face the issue with? `File => Save Project`

old-relic, 2021-11-28:
Here you go
I included the gcode my PC generates too.
Mike

old-relic, 2021-11-28:
If you examine the gcode file (i used notepad++) the last layer change
occurs at line 32845 in the file. Z=124.8. The box is actually a lot
taller than that.
FWIW
Mike

fe60, 2021-11-28:
@old-relic The project file is missing. I guess replying via mail doesn't upload attachments.

old-relic, 2021-11-28:
Trying again

[a123-battery-box.zip](https://github.com/prusa3d/PrusaSlicer/files/7614182/a123-battery-box.zip)

lukasmatena, 2021-11-28:
There is nothing wrong. The object is 125 mm tall. There are 417 layers generated, the last one is 125.1 mm above the print bed (line 32915). You are mixing up layer z and layer number. Closing.

closed 2021-11-28 by lukasmatena

old-relic, 2021-11-29:
Lukáš you are correct, i'm sorry for wasting your time.