prusa3d/PrusaSlicer #7361: Cannot send to printer - incorrect port

Shama71, 2021-11-27:
### Version
PrusaSlicer 2.4.0beta2

### Operating system type + version
Windows 11 64bit
### 3D printer brand / version + firmware version (if known)
### Behavior
Cannot send to printer - incorrect port.

[image]

#### Project File (.3MF) where problem occurs

matthewlloyd, 2021-11-27:
I think this bug was introduced a couple of days ago and happens only on Windows:

https://github.com/prusa3d/PrusaSlicer/commit/e276b7085126d7e348936765125dbb178037c12a#diff-e9db725725d583ec3c98227a2d312528cb758e6458018f82c566ff61070a46dbR80

The port information in the printer address is stripped out by the IP resolve process, but does not get added back in. As a result the test request to api/version goes to the specified port, but the subsequent api/files/local request now always goes to port 80.

Shama71, 2021-11-27:
My printer settings use IP address, but not hostname.

lukasmatena, 2021-11-28:
@Shama71  @matthewlloyd Thanks, we'll look into it.

napter, 2021-11-30:
Just encountered this - previous release works fine but beta2 now refuses to connect to octoprint on port 5000 and instead attempts to connect on port 80.

kocikdav, 2021-12-01:
Please try this [build](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.4.0-beta2+43-dk-printhost-win64-ge398289d3-202112010937.zip). (if prompted before download use login prusaslicer : slicer).  Also please note that this build is experimental and other features could be broken or in middle of development.

When you run PrusaSlicer with command line, there should be output (marked as error but its not an error) to log real address for upload.

Shama71, 2021-12-01:
Send to Octoprint was successful.

closed 2021-12-01 by kocikdav

kohend, 2021-12-04:
I think that closing issues that are fixed in master but not released may be confusing, maybe change it to pending or something similar?