prusa3d/PrusaSlicer #7363: Some Dark Theme UI Elements look out of place

NHodgesVFX, 2021-11-27:
### Version
Version 2.4.0-beta2+win64

### Operating system type + version
Windows 10 64Bit

### 3D printer brand / version + firmware version (if known)
Prusa Mini +

### Behavior
The Drop down boxes in the dark theme look out of place. Right now they have almost white borders surrounding them and the button off to the right side is also a brighter white, this does not match with the rest of the UI. Side note here as well the focus for the brim check box is broken, it appears to the side rather than around the element. Last Side note, for this section, the infill drop down is way thicker than any of the other drop downs.

[image]

Additionally, when opening the drop down the outline is blue compared to the rest of the UI which is grey / white.
[image]

The checkbox element also feels out of place do to the harsh white background, although this one doesn't work that bad with the dark theme. Something not related to the dark theme is the lock Icon for scaling would look nicer on the other side. This would make it so no indent is needed. Moving it to the other side would also make it consistent with the drop to bed icon on the position row.

[image]

Here is a look at the drop down from FreeCad, Also notice the scrollbar theme as well. 
[image]

Drop Down Open:
[image]

Check Boxes:
[image]

Here is substance painter
Drop Down:
[image]

Drop Down Open:
[image]

Check Boxes:
[image]

I am aware that UI is subjective but I wanted to leave what I thought would look nice. Thanks for the awesome software :)

_Is this a new feature request?_
Kindof

lukasmatena, 2021-11-28:
Thanks for the feedback. We will look at it, but I don't promise that we'll continue pushing dark mode to complete perfection. Because of internal architecture of PrusaSlicer and inadequate MS support, dark mode was about much more than magically switching it on. See the discussion in https://github.com/prusa3d/PrusaSlicer/issues/4493#issuecomment-803375331. @YuSanka has already spent many woman-days on it, and at one point we'll have to say it is good enough.

FreeCAD that you mentioned uses Qt for rendering the controls, so I guess that dark mode support was easier for them. Qt draws all the widgets itself, and as far as I know, they are skinnable (I don't know if Qt provides a dark mode palette out of the box). This is also the answer to why they don't look the same in PrusaSlicer, which uses a combination of undocumented dark mode Windows API and owner-drawn widgets. We can of course use it for inspiration, that's for sure.

bubnikv, 2021-11-29:
@NHodgesVFX you may check dark mode support in Notepad++. Notepad++ is written in Win32 and they spent a lot of effort into making it work in dark mode.

If you don't like PrusaSlicer's dark mode support, it is best to complain to Microsoft. We should gang up and knock their doors.

closed 2026-09-29 by prusaslicer-bot[bot]