prusa3d/PrusaSlicer #7366: "add part" error - PrusaSlicer 2.4.0 b2

stefanpieter, 2021-11-28:
### Version
- PrusaSlicer 2.4.0 b2

### Operating system type + version
Mac MackBook Air M1

### Behavior
When added a part using the "add part" function it only slices the last part added and not the entire component. I have tried different parts with the same issue.

https://imgur.com/a/1dMyxdB

lukasmatena, 2021-11-28:
Thanks, but I cannot reproduce it. Do you have precise steps to reliably and repeatedly reproduce it? The bug may depend on what you did before.

bubnikv, 2021-11-29:
The update code was refactored strongly since PrusaSlicer 2.3.3 due to better support of multi-material, multi-material painting etc, however the code is in a heavy use thus bugs are expected to be found quickly. We would be very thankful if you could help finding a reproducible scenario where the update code fails.

lukasmatena, 2021-11-29:
@stefanpieter I actually reproduced the problem by just loading and slicing the 3MF. This means we will be able to fix. Thanks a lot.

bubnikv, 2021-11-29:
Fixed with 6883919a7bbabf68830c19d5cc50f05367a2c604
    This was a regression of 2.4.0-alpha1 wrt. 2.3.3, where handling of
    multiple volumes was refactored.
    As a result, handling multi-part objects with clip_multipart_objects
    disabled was broken.
    
    With clip_multipart_objects disabled, the multi-part objects no more
    clip one another and it is sole responsibility of the user to handle
    overlaps.

closed 2021-11-29 by bubnikv