prusa3d/PrusaSlicer #7367: [bug] Support on undesirable places

batata004, 2021-11-28:
Take a look at this part:

[image]

I painted the support enforcer only in the hanging part but it came out like this:

[image]

You can clearly see that there is support material supporting absolutely nothing. The support should be only under the hanging part, but it also "leaks" to the side and for some strange reason support is being generated directly in the build plate, supporting nothing.

My support configuration is exactly this:

[image]

If there is some way I can fix this on my own instead of waiting this bug to be fixed, could you please tell me? Because this happens a lot (this happened for a long time, but I know that the latest version received many support improvements so I decided to report this hoping someone is willing to fix this).

lukasmatena, 2021-11-28:
Please, post a 3MF and tell us what PrusaSlicer version you are talking about. Thanks.

batata004, 2021-11-28:
@lukasmatena I am attaching you the file you requested. I am using the latest stable version (not rc).
[File.zip](https://github.com/prusa3d/PrusaSlicer/files/7614484/File.zip)
.

lukasmatena, 2021-11-28:
@batata004 
> I am using the latest stable version (not rc).

Last stable is 2.3.3 and the config screenshot you posted contains options that were not present there. The 3MF that you posted says it was generated in 2.4.0-beta1, so that's what I will trust.

I cannot tell if this is an issue and why is it there. The behaviour seems to be the same as it was in 2.3.3, so it is not anything that would need an immediate fix (I'm mentioning this because 2.4.0 final is close). You can try the new "Snug" supports (from the "Style" dropdown). This is a new option in 2.4.0 and might give you better result in this case.

lukasmatena, 2021-11-28:
You can also remove the paint from the very edge of the overhang so there is an unpainted margin. That is what is causing the problem.

batata004, 2021-11-28:
Sorry, I have 3 computers and I sent you the file from the other computer. Anyway, I tried this in the other notebook (running the latest stable) and the bug still happens.

I respectfully dont agree with what you said "_I cannot tell if this is an issue_". Of course this is a bug, it does not happen on Cura and S3D. If this does not convice you, I will give you 2 more reasons so you know this is surely a bug, 100% sure:

1) a support has a purpose which is support something. The support on the build plate is supporting absolutely nothing. It wastes material, time, energy AND the external surface of the part gets a bad texture (not smooth) because the unneeded support in the external part.

2) when I painted the support, I made it very clear WHERE the support should be. However, as you can see from my print, I told support to be just under the overhang but PS placed the support there an in another part that I didnt explicitly paint.

This bug happens with many other parts, not only in this specific STL object. If this is not a bug, I must have forgotten the definition of a bug.

Anyway, I provided you my time and effort to come here, give you prints, files... so I could help the community improve this software pointing some bugs. If you or anyone else has no interest solving this bug, claiming it's not a bug not even an issue, it just means I wasted my time and I will surely not make this mistake again with PS.

lukasmatena, 2021-11-29:
@batata004 Calm down, I'm not saying it does not exist. But yes, there is a difference between a bug (algorithm not working as designed) and an algorithm working as designed but not providing a result people expect. I have a feeling that what you posted is an inherent property of Grid supports, which work on a regular grid and thus tend to leak into neighboring regions as a cost for increased and guaranteed stability. Snug supports introduced in 2.4.0 were provided as an alternative, the situation is described in the [2.4.0-alpha1 change log](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1), with several similar issue tickets refrenced from there (in *FDM supports improvements* paragraph).

@bubnikv Can you please check that what I wrote applies here? Thanks.

bubnikv, 2021-11-29:
The "snug" support style in PrusaSlicer 2.4.0-beta2 surely solves the supports expansion.

[image]

Closing.

closed 2021-11-29 by bubnikv

batata004, 2021-11-29:
This bug is still happening even using SNUG. You can see there is a dark green line outside of the part overhang. It does not reach the build plate but it's outside of the overhang. Even in the print you provided when you said "Closing." you can clearly see the bug: check the dark green lines, they are outside of the overhanging region.
[1.zip](https://github.com/prusa3d/PrusaSlicer/files/7618764/1.zip)