prusa3d/PrusaSlicer #7368: Long Custom Code causes Tooltip/preview flicker that blocks right-click for Edit etc

Jaxx005, 2021-11-28:
### Version
Version 2.3.3+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S standard, current latest firmware.

### Behavior
Tooltip (mouse-over preview) of Custom Code (beyond certain length) flickers and can impeded or prevent right-click menu and hence Edit.

1. Place Custom Code on a layer (more central layers tested)
2. Select that layer to display it
3. Hover mouse over that layer marker (+)
4. The Custom Code is displayed in a pop-up 'Tooltip' box... [normal = no flicker]
5. Right-clicking the layer marker is possible, plus Edit etc. [normal]
6. Alter the custom code to be (?) 27 or more lines long (I used ";" on each line)
7. Repeating (4) above, shows a rapidly flickering 'Tooltip' box
8. Right-clicking (5) is difficult or impossible, as so therefore is selecting 'Edit...' etc.

This is a bug it seems:
- Survived a restart and system reboot.  
- Repeatable using a new PS project, with just a primary 'block' object added etc.

#### Project File (.3MF) where problem occurs

[Custom GCode - Tooltip Flicker Example.zip](https://github.com/prusa3d/PrusaSlicer/files/7612819/Custom.GCode.-.Tooltip.Flicker.Example.zip)

ZIP file also includes PDF with screenshot (was going more than 1!)

This is probably quite an obscure 'bug', but a very frustrating one that possibly varies with the specification of the users PC/GPU perhaps?   I don't think it's installation-specific, but I supposed it could be?

Thanks. TTFN.

lukasmatena, 2021-11-28:
Thanks, we'll investigate.

lukasmatena, 2021-11-30:
Should be fixed with b164271. Closing.

closed 2021-11-30 by lukasmatena

Jaxx005, 2021-12-02:
Looks like you found an existing post!  Good find!