prusa3d/PrusaSlicer #7369: disappearing stuff

jmerc77, 2021-11-28:
### Version
Version 2.4.0-beta2+win64

### Operating system type + version
windows 10

### 3D printer brand / version + firmware version (if known)
cr 10s (but not a printing issue)

### Behavior
icons, buttons, etc. vanished mid setting of print settings. got so bad it would not save even if I clicked the now invisible save icon and restarted prusa slicer. also dialogs (such as: save, closing dialogs, probably more) were too big and empty. this was the first time running this version. not sure if it is a setup bug. also profiles were imported from v2.3.3 stable before attempting to make changes.

did notice a brief usage of my hdd before it happened. also not running from the system drive (put on another drive instead of C drive)

it happened seemingly randomly.
stuff shouldn't disappear at random
lots of things just randomly started to vanish

#### Project File (.3MF) where problem occurs
none no objects loaded; first time use.

fe60, 2021-11-28:
Any chance to create a screenshot of the corrupt UI?
Can you try running PrusaSlicer with logging and share the output if there is something that seems to be related to the strange behaviour? (see comment from developer in [#4316 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/4316#issuecomment-634769532))

jmerc77, 2021-11-28:
I don't have a screenshot. the issue seems to not be repeatable as I haven't seen the problem since that first run. also where would the logs be stored? maybe there are logs that I don't know about.

jmerc77, 2021-11-28:
just happened again! also clipboard was glitchy and had a hard time getting the screen shots but was able to
[image]
[image]
.

jmerc77, 2021-11-28:
hay! 21h2 was not installed on my pc. maybe things will clean up on my end after update (like the clipboard).

jmerc77, 2021-11-28:
ummm just got this error after windows update:

Most likely the configuration was produced by a newer version of PrusaSlicer or by some PrusaSlicer fork. The following values were substituted:

Print settings : <b>"0.2 raft"</b> (C:\Users\Owner\AppData\Roaming\PrusaSlicer-beta\print\0.2 raft.ini)<table><tr><td><b>"support_material_interface_pattern" (Interface pattern)</b></td><td>: <b>"rectilinear-grid"</b> was substituted with <b>"auto" (Default)</b></td></tr></table>
Print settings : <b>"CR-10S"</b> (C:\Users\Owner\AppData\Roaming\PrusaSlicer-beta\print\CR-10S.ini)<table><tr><td><b>"support_material_interface_pattern" (Interface pattern)</b></td><td>: <b>"rectilinear-grid"</b> was substituted with <b>"auto" (Default)</b></td></tr></table>
Review the substitutions and adjust them if needed.

n8bot, 2021-11-29:
This rectilinear grid support material interface error message is what I was targeting for a fix with https://github.com/prusa3d/PrusaSlicer/pull/7316

bubnikv, 2021-11-29:
The "support_material_interface_pattern" fix by @n8bot was merged, thanks.
Regarding the main issue with blanking interface, I don't know. How much RAM is your computer equipped with? Are you slicing large models?

jmerc77, 2021-11-30:
16gb. no models were loaded or sliced. just initial setup of my settings. also if i launch it fresh and load a 3dbenchy and slice, that is working. its when i stay in the setting too long.

jmerc77, 2021-11-30:
in case of bad values in config here's my config snapshot:
[config snap.zip](https://github.com/prusa3d/PrusaSlicer/files/7622821/config.snap.zip)

jmerc77, 2021-11-30:
opened it up today with my phone stop watch running... 30 minutes in and no issues yet even though i was poking around a bit. still open as i press coment at 33 min

jmerc77, 2021-11-30:
while i am not sure the cause, I'm thinking it was pending windows updates because of the weird clipboard crash that happened at the same time. since i cannot seem to repeat it by poking arround, i am closing this until i notice it again.

n8bot, 2021-12-05:
I actually had this problem too, right now, on latest master branch.

I was working on a new set of profiles for another user, so I was making lots of changes and saving a lot. The crazy corruption/disappearing of elements in the UI occurred when I saved the print profile, and quickly changed to a new settings page immediately after setting.

I'm not sure how reproducible that will be, but it was nuts for sure...

n8bot, 2021-12-05:
As an added piece of info, I now confirm that the save did not take effect. I clicked save profile, accepted the dialog, then quickly changed the settings page. The crazy UI stuff happened, I tried to quit, the dialogs were full screen but nearly blank, then the program crashed and stuff. Loaded up, the save that I initiated was not effective, the profile is as it was before I clicked save.

jmerc77, 2021-12-05:
was this on window 10 21h2 or was it an older update / version? because so far I haven't seen it yet after windows update to 21h2. but while I am working on a 3d printing project using prusa slicer, I haven't been in the settings very long though. I did find another annoying bug in modifiers though. that one is recoverable but requires starting fresh, starting over and resaving.

bubnikv, 2021-12-05:
>   did find another annoying bug in modifiers though. that one is
recoverable but requires starting fresh, starting over and resaving.

Description on how to reproduce would be great.

jmerc77, 2021-12-05:
more on that here. since it is not relevant to this issue:
#7429

n8bot, 2021-12-05:
> was this on window 10 21h2 or was it an older update / version? because so far I haven't seen it yet after windows update to 21h2.

In case this does mater, I'm on 20H2.

lukasmatena, 2021-12-06:
@n8bot It apparently crashed inside std::string copy constructor after being passed a rotten reference, which can mean just about anything, especially if it crashed some time after the initial corruption took place. If you happened to reproduce it, we would be glad if you could capture a complete call stack at the time of the crash.

n8bot, 2021-12-06:
Yeah, now that you mention it, the crash took place after I "clicked yes" on a misshapen dialog box that was essentially full screen, grey, with the orange words "yes." The "dialog" popped up when I tried to close the app by pressing the windows "X" at the top right title bar. 

I will try to reproduce it, but I already tried yesterday and couldn't get it.

n8bot, 2021-12-06:
Well, I corrupted the settings (finally!) and then the app crashed when I was just moving around the settings pages. Everything is/was not quite as corrupt as the other time, but similar.

Unfortunately, this was done on my customized version and sadly when I was switching one of the settings that I have added myself to the app (the optimize tool order setting). So, we obviously can't rule out that this is my problem here...

BUT, the first time I corrupted the settings, yesterday, it was not with one of my custom settings. And, afaik, the original user was not using my modified version.

Anyway:
[image]

```
Exception thrown at 0x00007FFBCFE476EA (PrusaSlicer.dll) in prusa-slicer-console.exe: 0xC0000005: Access violation reading location 0x0000000000000008.
```

Call stack:
```
 	PrusaSlicer.dll!wxObject::Ref(class wxObject const &)	C++
 	PrusaSlicer.dll!wxTextMeasureBase::DoGetPartialTextExtents(class wxString const &,class wxArrayInt &,double)	C++
 	PrusaSlicer.dll!wxTextMeasure::DoGetPartialTextExtents(class wxString const &,class wxArrayInt &,double)	C++
 	PrusaSlicer.dll!wxTextMeasureBase::GetPartialTextExtents(class wxString const &,class wxArrayInt &,double)	C++
 	PrusaSlicer.dll!wxMSWDCImpl::DoGetPartialTextExtents(class wxString const &,class wxArrayInt &)	C++
 	PrusaSlicer.dll!wxTextWrapper::Wrap(class wxWindow *,class wxString const &,int)	C++
 	PrusaSlicer.dll!wxStaticTextBase::Wrap(int)	C++
>	PrusaSlicer.dll!Slic3r::GUI::ogStaticText::SetText(const wxString & value, bool wrap) Line 1030	C++
 	PrusaSlicer.dll!Slic3r::GUI::TabPrinter::update_machine_limits_description(const Slic3r::MachineLimitsUsage usage) Line 3974	C++
 	PrusaSlicer.dll!Slic3r::GUI::TabPrinter::toggle_options() Line 3019	C++
 	PrusaSlicer.dll!Slic3r::GUI::Tab::activate_selected_page(std::function<void __cdecl(void)> throw_if_canceled) Line 3496	C++
 	PrusaSlicer.dll!Slic3r::GUI::TabPrinter::activate_selected_page(std::function<void __cdecl(void)> throw_if_canceled) Line 2915	C++
 	PrusaSlicer.dll!Slic3r::GUI::Tab::tree_sel_change_delayed() Line 3554	C++
 	[Inline Frame] PrusaSlicer.dll!Slic3r::GUI::Tab::create_preset_tab::__l2::<lambda_4c2cb291d99f0d4876f0a605c407bdc3>::operator()(wxTreeEvent &) Line 336	C++
 	PrusaSlicer.dll!wxEventFunctorFunctor<wxEventTypeTag<wxTreeEvent>,<lambda_4c2cb291d99f0d4876f0a605c407bdc3>>::operator()(wxEvtHandler * __formal, wxEvent & event) Line 536	C++
 	[External Code]	
 	PrusaSlicer.dll!Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams & params) Line 54	C++
 	PrusaSlicer.dll!Slic3r::CLI::run(int argc, char * * argv) Line 618	C++
 	PrusaSlicer.dll!slic3r_main(int argc, wchar_t * * argv) Line 838	C++
 	prusa-slicer-console.exe!wmain(int argc, wchar_t * * argv) Line 311	C++
 	[External Code]	
```
[image]

n8bot, 2021-12-06:
I got it to crash again, with similar steps: changing settings and saving for like 20+ minutes. eventually it happens. This time, with support setting change (increased raft layers to 1)I was previously messing around in the printer settings page, which seems to be the root of this problem (Machine limits usage page? Similar to the problem with linux and setText()?):

```
Exception thrown at 0x00007FFBCFCEC686 (PrusaSlicer.dll) in prusa-slicer-console.exe: 0xC0000005: Access violation reading location 0x0000000000000000.
```

Call stack:
```
 	PrusaSlicer.dll!wxSpinCtrl::DoMoveWindow(int,int,int,int)	C++
 	PrusaSlicer.dll!wxWindow::DoSetSize(int,int,int,int,int)	C++
>	[Inline Frame] PrusaSlicer.dll!wxWindowBase::Move(int) Line 273	C++
 	[Inline Frame] PrusaSlicer.dll!wxWindowBase::Move(const wxPoint &) Line 276	C++
 	[Inline Frame] PrusaSlicer.dll!wxWindowBase::SetPosition(const wxPoint &) Line 278	C++
 	PrusaSlicer.dll!Slic3r::GUI::OG_CustomCtrl::correct_window_position(wxWindow * win, const Slic3r::GUI::Line & line, Slic3r::GUI::Field * field) Line 352	C++
 	PrusaSlicer.dll!Slic3r::GUI::OG_CustomCtrl::CtrlLine::correct_items_positions() Line 475	C++
 	PrusaSlicer.dll!Slic3r::GUI::OG_CustomCtrl::CtrlLine::update_visibility(Slic3r::ConfigOptionMode mode) Line 528	C++
 	PrusaSlicer.dll!Slic3r::GUI::OG_CustomCtrl::update_visibility(Slic3r::ConfigOptionMode mode) Line 335	C++
 	PrusaSlicer.dll!Slic3r::GUI::ConfigOptionsGroup::update_visibility(Slic3r::ConfigOptionMode mode) Line 676	C++
 	PrusaSlicer.dll!Slic3r::GUI::Page::activate(Slic3r::ConfigOptionMode mode, std::function<void __cdecl(void)> throw_if_canceled) Line 4083	C++
 	PrusaSlicer.dll!Slic3r::GUI::Tab::activate_selected_page(std::function<void __cdecl(void)> throw_if_canceled) Line 3494	C++
 	PrusaSlicer.dll!Slic3r::GUI::Tab::tree_sel_change_delayed() Line 3554	C++
 	[Inline Frame] PrusaSlicer.dll!Slic3r::GUI::Tab::create_preset_tab::__l2::<lambda_4c2cb291d99f0d4876f0a605c407bdc3>::operator()(wxTreeEvent &) Line 336	C++
 	PrusaSlicer.dll!wxEventFunctorFunctor<wxEventTypeTag<wxTreeEvent>,<lambda_4c2cb291d99f0d4876f0a605c407bdc3>>::operator()(wxEvtHandler * __formal, wxEvent & event) Line 536	C++
 	[External Code]	
 	PrusaSlicer.dll!Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams & params) Line 54	C++
 	PrusaSlicer.dll!Slic3r::CLI::run(int argc, char * * argv) Line 618	C++
 	PrusaSlicer.dll!slic3r_main(int argc, wchar_t * * argv) Line 838	C++
 	prusa-slicer-console.exe!wmain(int argc, wchar_t * * argv) Line 311	C++
 	[External Code]	
```

[image][image][image]

bubnikv, 2021-12-07:
The disappearing windows should be fixed with 
https://github.com/prusa3d/wxWidgets/commit/296b2b4d08b53c6136de400acbb1d826ed4e2af8

Please help to test this build:
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta3+18-win64-g89eabec58-202112071305.zip
You may need to fill in a login dialog if it opens:
user prusaslicer
password slicer

thank you.

closed 2021-12-07 by bubnikv

n8bot, 2021-12-07:
I will test it when I finish waking up. 

Can I compile my own, if I recompile deps? Or is your patched wxWidgets not yet included?

bubnikv, 2021-12-07:
> Can I compile my own, if I recompile deps?

Sure.

> Or is your patched wxWidgets not yet included?

The commit referenced 
https://github.com/prusa3d/wxWidgets/commit/296b2b4d08b53c6136de400acbb1d826ed4e2af8
is in.

n8bot, 2021-12-07:
I'm just getting ready to test it now. But, the original user who reported this was not using dark mode. Do you think the problem originated in dark mode, but also presented itself in light mode?

bubnikv, 2021-12-07:
Yes, the problem happened in light mode as well.

n8bot, 2021-12-07:
Cool cool. I built the deps, confirming (because I'm a newb) that wxwidgets was the new patched version (it was, thanks!) and I'm now rebuilding the app. IMO, if I can't reproduce the corruption with 30 mins of saving random config changes, I will call it fixed.

n8bot, 2021-12-07:
I confirm this seems fixed. Over the course of an hour, I saved countless random settings changes and never got corruption. Thanks!

bubnikv, 2021-12-08:
Thanks a lot for testing.