prusa3d/PrusaSlicer #7371: Hello. The problem of transferring g-code to octoprint. PrusaSlicer 2.4.0 beta2

ser159, 2021-11-28:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

ser159, 2021-11-28:
PrusaSlicer
Version:   2.4.0-beta2+win64
Build:     PrusaSlicer-2.4.0-beta2+win64-202111261611

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 6,314MB
OpenGL installation
GL version:   4.6.148201 Compatibility Profile Context 20.50.02.05 27.20.15002.5003
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon(TM) Graphics
GLSL version: 4.60
<details>
<summary>Installed extensions:</summary>
GL_AMDX_debug_output
GL_AMD_blend_minmax_factor
GL_AMD_conservative_depth
GL_AMD_debug_output
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_sample_positions
GL_AMD_gcn_shader
GL_AMD_gpu_shader_half_float
GL_AMD_gpu_shader_half_float_fetch
GL_AMD_gpu_shader_int16
GL_AMD_gpu_shader_int64
GL_AMD_interleaved_elements
GL_AMD_multi_draw_indirect
GL_AMD_name_gen_delete
GL_AMD_occlusion_query_event
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_sample_positions
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_atomic_counter_ops
GL_AMD_shader_stencil_export
GL_AMD_shader_stencil_value_export
GL_AMD_shader_trace
GL_AMD_shader_trinary_minmax
GL_AMD_sparse_texture
GL_AMD_sparse_texture_pool
GL_AMD_stencil_operation_extended
GL_AMD_texture_cube_map_array
GL_AMD_texture_texture4
GL_AMD_transform_feedback3_lines_triangles
GL_AMD_transform_feedback4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
[... 226 lines cut ...]
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KTX_buffer_region
GL_NV_alpha_to_coverage_dither_control
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_explicit_multisample
GL_NV_float_buffer
GL_NV_half_float
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_OES_EGL_image
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

sg000, 2021-11-28:
Not sure the OP did a great job explaining the situation that causes the issue, but I have this error as well. I'm jumping from 2.3 to 2.4 Beta2, but I get this error on GCODE save. If I remove my Python post-process line "****/python.exe ****/script.py" it saves ok. When that call is in there, the python gets called, but then it I get a Prusa pop-up window telling me copying the temporary G-Code output failed.

[image]

Windows 64-bit.

This is the first slice I've tried with Beta2, so my settings are all brought over from the 2.3 version I previously used.

neophyl, 2021-11-28:
There were changes to the post processing in the 2.4 alpha.  This post might also contain some useful information https://github.com/prusa3d/PrusaSlicer/issues/7190

pauluchin, 2021-11-28:
Same error here...
try to upload the gcode i have it set to port 5000 (checked in case I made any mistakes)
[image]

ser159, 2021-11-28:
Сonnect to port 80, I have 5000. Which is wrong. While I'm sitting on version 2.3.3

lukasmatena, 2021-11-28:
@ser159 @pauluchin Your issue is likely a duplicate of #7361.

@sg000 Your issue is likely connected to #7190, as @neophyl suggested. If so, please, comment on that issue.

Duplicate, closing.

closed 2021-11-28 by lukasmatena