prusa3d/PrusaSlicer #7372: Tooltips invisible in Beta2.4.0

bbinnard, 2021-11-28:
### Version
2.4.0 Beta2

### Operating system type + version
Windows11/64 Home with Dark theme

### Behavior
Tooltips are white on white - text is invisible

[image]

lukasmatena, 2021-11-28:
Thanks, we'll try to reproduce and fix it.

YuSanka, 2021-11-30:
@bbinnard, do you have some special settings for the color mode of your Windows?
We can't reproduce your issue or regular Dark and Light mode
[image]

bbinnard, 2021-11-30:
This problem is fixed by choosing a different theme in the Windows 11 Personalization/Themes section. In the attached screenshot the red arrow points to the theme that causes the problem, and the green arrow points to one of the themes that eliminates it. There are probably other themes that behave the same way, but I didn't check for that.

[image]

closed 2021-11-30 by bbinnard