prusa3d/PrusaSlicer #7374: Make the Ender 3V2 bed **texture** the same as the normal ender 3

marconovino, 2021-11-29:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

The ender 3 v2 bed is very bland and a blank bed, and its technically innacurate since the V2 comes with a glass bed that says Ender and has the dragon logo, just like the original ender 3. The ender 3 v2 texture is unnecessary, just use the normal ender 3 one instead since its the real one

rtyr, 2021-11-29:
I don't have a strong opinion about this. The bed model doesn't have to be technically accurate and dragon logo in the middle may look cool, but it can also obstruct the view from the bottom.

FYI @pmjdebruijn

pmjdebruijn, 2021-12-04:
Yeah, practically it's mostly a detriment to having some illustration on the bed. When I can find some time I'll likely will be removing the texture for the regular Ender-3 as well. That way all the Creality printers will be consistent.

closed 2026-09-29 by prusaslicer-bot[bot]