prusa3d/PrusaSlicer #7375: M1 Gcode

PompTel, 2021-11-29:
### Version
2.4.0 beta2 win64

_Use `About->About Prusaslicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
windows11

### 3D printer brand / version + firmware version (if known)
Geeetech a20m

### Behaviour

The gcode is including an M1 code when i try to print a multicolour print before it prints the model, but after it does the wipe

[image]

_Is this a new feature request?_
No

Hi i dont want the printer to pause every time i create multicolour print, im using Octoprint and Octoprint tells me this is a blocked command?  im relatively new to printing and just moved over to Prusaslicer as i think its better than Cura, but unsure if i can remove the blocked command from Octoprint, or if there's a way of me removing it within Prusaslicer settings so that it no longer adds an M1 code in future?

lukasmatena, 2021-11-29:
The `M1` after the priming extrusions (at least I assume that's what you're talking about, since there is not 3MF attached) is currently hardcoded for Marlin. It should be a valid Marlin command, the other firmware flavors do not emit it (see also #5441). I don't know why would OctoPrint reject it and I don't remember this would be ever reported. Can you please post a 3MF?

As a workaround, you can either disable 'Prime all printing extruders' (Print Settings->Multiple extruders) or set firmware flavor to RepRapFirmware (there are minimal differences with respect to Marlin).

PompTel, 2021-11-29:
hello again, thankyou for the speedy reply.  im sorry i really dont mean to sound like the thick one here, but im still feeling my feet with various aspects of 3d printing/slicing etc.  can you let me know what a 3mf file is please? and ill be happy to oblige

lukasmatena, 2021-11-29:
I updated my previous post and changed `RepRapFirmware` to `RepRap/Sprinter`. That should be the correct alternative. Thanks @rtyr for noticing that.

3MF is a project file format that PrusaSlicer (and others) use for storing models along with custom metadata. When you set your plater in PrusaSlicer and set some settings, you can save your project (File->Save As). This creates a 3MF file containing both the geometry and settings, making it easy for anyone to reproduce exactly what you are doing.

PompTel, 2021-11-29:
OK thankyou that makes sense and sorry for not understanding initially.  
Im off to work now, so will return to this later with the relevant file, and i'll also take M1 out of the blocked list on Octoprint, as you say its used ok everywhere else, and this may just be an unnecessary blocked command

PompTel, 2021-11-30:
Switched to RepRap/Splinter as suggested and it resolved the issue.  thanks

closed 2021-11-30 by PompTel