prusa3d/PrusaSlicer #7376: 2.4.0-beta2 crash when viewing sliced gcode

Pitel, 2021-11-29:
### Version
Verze 2.4.0-beta2+linux-x64-GTK3

### Operating system type + version
Ubuntu 21.10

PrusaSlicer
Version:   2.4.0-beta2+linux-x64-GTK3
Build:     PrusaSlicer-2.4.0-beta2+linux-x64-GTK3-202111261605

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.13.0-22-generic x86_64
Total RAM size [MB]: 16,727MB
OpenGL installation
GL version:   4.6.0 NVIDIA 495.44
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce GTX 1060 3GB/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
[... 338 lines cut ...]
GL_SGIX_shadow
GL_SUN_slice_accum
</details>

### 3D printer brand / version + firmware version (if known)
Prusa Mini (rev2, not plus), firmawe 4.3.4-RC

### Behavior
1. Add STL file mentioned later
2. Slice it
3. Try to switch to GCode viewer tab
4. Crash :boom:

```
Unhandled unknown exception; terminating the application.
terminate called without an active exception
[1]    31481 IOT instruction (core dumped)  /opt/PrusaSlicer-2.4.0-beta2+linux-x64-GTK3-202111261605.AppImag
```

* Slicing default cube is ok.
* beta1 works fine

#### Project File (.3MF) where problem occur
* [Hyrule_Crest_v2.zip](https://github.com/prusa3d/PrusaSlicer/files/7617868/Hyrule_Crest_v2.zip)
* https://www.prusaprinters.org/cs/prints/84138-slim-raspberry-pi-zero-2-w-case/files (STLs)

bubnikv, 2021-11-29:
Would you please attach a 3MF? The 3MF contains all the parameters we need to reproduce your issue. Thank you.

Pitel, 2021-11-29:
[3MF in ZIP](https://github.com/prusa3d/PrusaSlicer/files/7618119/Hyrule_Crest_v2.zip)

bubnikv, 2021-11-29:
I am sorry I cannot reproduce your issue. Anybody else could reproduce this crash?

bubnikv, 2021-11-29:
However this "Linux G-code viewer" smells after a bad OpenGL driver to me.

Pitel, 2021-11-29:
Well, it might be. But beta1 works fine, 2.3.3 works fine, Blender works fine...

Pitel, 2021-11-30:
No alacritty terminal emulator, also no Snaps.

X11. Downloaded from GitHub (GTK3 version).

```
[2021-11-30 12:21:51.976131] [0x00007fc3defad8c0] [trace]   Initializing StaticPrintConfigs
[2021-11-30 12:21:52.053474] [0x00007fc3defad8c0] [debug]   full path: 901560929177738410
[2021-11-30 12:21:52.053501] [0x00007fc3defad8c0] [debug]   single instance: undefined. other params: /tmp/.mount_PrusaSMbcYPL/usr/bin/bin/prusa-slicer;--loglevel;5
[2021-11-30 12:21:52.053505] [0x00007fc3defad8c0] [debug]   full lock path: /home/pitel/.config/PrusaSlicer-beta/cache/901560929177738410.lock
[2021-11-30 12:21:52.053520] [0x00007fc3defad8c0] [debug]   Creating lockfile.
[2021-11-30 12:21:52.053523] [0x00007fc3defad8c0] [info]    Instance check: Another instance not found or single-instance not set.
[2021-11-30 12:21:52.125584] [0x00007fc3defad8c0] [trace]   System language detected (user locales and such): cs_CZ
[2021-11-30 12:21:52.127953] [0x00007fc3defad8c0] [trace]   Best translation language detected (may be different from user locales): cs_CZ
[2021-11-30 12:21:52.127971] [0x00007fc3defad8c0] [trace]   Switching wxLocales to cs_CZ
[2021-11-30 12:21:52.199480] [0x00007fc3defad8c0] [info]    Checking if indices need to be installed from resources...
[2021-11-30 12:21:52.548047] [0x00007fc3defad8c0] [info]    Undo / Redo snapshot taken: Nový Projekt, Undo / Redo stack memory: 0MB Resident memory: 107MB; Shared memory: 61MB; Private memory: 46MB; Peak memory usage: 107MB
[2021-11-30 12:21:52.549056] [0x00007fc3b77fe640] [trace]   Dbus object /com/prusa3d/prusaslicer/InstanceCheck/Object901560929177738410 registered. Starting listening for messages.
[2021-11-30 12:21:52.602737] [0x00007fc3defad8c0] [debug]   BuildVolume bed_shape clasified as: Rectangle
[2021-11-30 12:21:52.927874] [0x00007fc3defad8c0] [debug]   TriangleMesh::repair() started
[2021-11-30 12:21:52.928371] [0x00007fc3defad8c0] [debug]   TriangleMesh::repair() finished
[2021-11-30 12:21:53.760648] [0x00007fc3defad8c0] [info]    Checking for cached configuration updates...
[2021-11-30 12:21:53.760684] [0x00007fc3defad8c0] [info]    Confing bundle not installed for vendor FLSun, skipping: 
[2021-11-30 12:21:53.760698] [0x00007fc3defad8c0] [info]    Confing bundle not installed for vendor Voron, skipping: 
[2021-11-30 12:21:53.760707] [0x00007fc3defad8c0] [info]    Confing bundle not installed for vendor INAT, skipping: 
[2021-11-30 12:21:53.760716] [0x00007fc3defad8c0] [info]    Confing bundle not installed for vendor Artillery, skipping: 
[2021-11-30 12:21:53.760725] [0x00007fc3defad8c0] [info]    Confing bundle not installed for vendor TriLAB, skipping: 
[2021-11-30 12:21:53.760735] [0x00007fc3defad8c0] [info]    Confing bundle not installed for vendor BIBO, skipping: 
[2021-11-30 12:21:53.760745] [0x00007fc3defad8c0] [info]    Confing bundle not installed for vendor Ultimaker, skipping: 
[2021-11-30 12:21:53.760753] [0x00007fc3defad8c0] [info]    Confing bundle not installed for vendor gCreate, skipping: 
[2021-11-30 12:21:53.760762] [0x00007fc3defad8c0] [info]    Confing bundle not installed for vendor Creality, skipping: 
[2021-11-30 12:21:53.771517] [0x00007fc3defad8c0] [debug]   Vendor: Prusa Research, version installed: 1.4.0-beta2, version cached: 1.4.0-beta2
[2021-11-30 12:21:53.771546] [0x00007fc3defad8c0] [info]    Confing bundle not installed for vendor LulzBot, skipping: 
[2021-11-30 12:21:53.771557] [0x00007fc3defad8c0] [info]    Confing bundle not installed for vendor Anycubic, skipping: 
[2021-11-30 12:21:53.771563] [0x00007fc3defad8c0] [info]    No configuration updates available.
[2021-11-30 12:21:53.788903] [0x00007fc3ccaa8640] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version2`
*   Trying 185.115.1.124:443...
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* ALPN, offering http/1.1
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: CN=files.prusa3d.com
*  start date: Nov  9 19:56:02 2021 GMT
*  expire date: Feb  7 19:56:01 2022 GMT
*  subjectAltName: host "files.prusa3d.com" matched cert's "files.prusa3d.com"
*  issuer: C=US; O=Let's Encrypt; CN=R3
*  SSL certificate verify ok.
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version2 HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.4.0-beta2+linux-x64-GTK3
Accept: */*

* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Server: nginx
< Date: Tue, 30 Nov 2021 11:21:53 GMT
< Content-Length: 23
< Connection: keep-alive
< Last-Modified: Tue, 30 Nov 2021 11:00:11 GMT
< ETag: "17-5d1ff7618ef6c"
< Accept-Ranges: bytes
< 
* Connection #0 to host files.prusa3d.com left intact
[2021-11-30 12:21:53.916218] [0x00007fc3ccaa8640] [info]    Got PrusaSlicer online version: `2.3.3`. Sending to GUI thread...
[2021-11-30 12:21:53.916247] [0x00007fc3ccaa8640] [info]    Got PrusaSlicer online version: `2.4.0-beta2`. Sending to GUI thread...
[2021-11-30 12:21:53.917162] [0x00007fc3ccaa8640] [info]    Syncing configuration cache
[2021-11-30 12:21:53.917178] [0x00007fc3ccaa8640] [warning] No such vendor: FLSun
[2021-11-30 12:21:53.917183] [0x00007fc3ccaa8640] [warning] No such vendor: Voron
[2021-11-30 12:21:53.917188] [0x00007fc3ccaa8640] [warning] No such vendor: INAT
[... cut ...]
(The empty lines at the end delimits the actual click on the gcode view button.)

I might be able to build it myself and do a bisect, but no promises, my time is limited because family and life.

Also, this morning, first start of the Slicer, and it was working flawlessly. Now, it's crashing again. So I;m starting to lean more to the side that there is something wrong with my system. I'll try it with older (470) nVidia drivers tomorrow and I'll see if it makes any difference. I'll also try non-GTK3 (what is it then, anyway? GTK2? 4? Qt?) build right away.

Pitel, 2021-11-30:
FYI, non-GTK3 build is also crashing.

lukasmatena, 2021-11-30:
Thanks for the info.

> I might be able to build it myself and do a bisect, but no promises, my time is limited because family and life.

Totally understand. Issues like this may also be triggered by quite unrelated changes, so there is no actual guarantee that bisecting to the first bad commit actual helps. Trying with older drivers is a good idea.

Pitel, 2021-12-02:
It was working fine with 470 drivers the whole couple ofdays, but now it started crashing again. So, it's not drivers.

Pitel, 2021-12-02:
I wanted to bisect, but tag `version_2.4.0-beta2` has some bad wxWidgets hash, and `master` has this error:
```
PrusaSlicer/src/slic3r/GUI/OptionsGroup.cpp:992:17: error: ‘format_wxstr’ was not declared in this scope
  992 |                 format_wxstr(_L("Visit \"Preferences\" and check \"%1%\"\nto changes your choice."), preferences_item);
      |                 ^~~~~~~~~~~~
```

hejllukas, 2021-12-02:
> I wanted to bisect, but tag `version_2.4.0-beta2` has some bad wxWidgets hash, and `master` has this error:
>
> ```
> PrusaSlicer/src/slic3r/GUI/OptionsGroup.cpp:992:17: error: ‘format_wxstr’ was not [...]

This compilation error should be fixed in the current `master` using f88d678a4a8957377f0145749e594d67225527cf.

Pitel, 2021-12-03:
Alright, `master` is not crashing 👍🏻, but how can I build `version_2.4.0-beta2` tag when the hash check fails?

bubnikv, 2021-12-03:
hash check should not fail anymore

Pitel, 2021-12-03:
It still does.

```
-- verifying file...
       file='/home/pitel/PrusaSlicer/deps/build/wxWidgets/v3.1.4-patched.zip'
-- SHA256 hash of
    /home/pitel/PrusaSlicer/deps/build/wxWidgets/v3.1.4-patched.zip
  does not match expected value
    expected: '21ed12eb5c215b00999f0374af652be0a6f785df10d18d0dfec8d81ed4abaea3'
      actual: 'ed36a2159c781cce07b06378664e683ebd8cb2f51914aba9acd3bfca3d63d7d3'
-- File already exists but hash mismatch. Removing...
```

I'm on commit `2c5ddfef` (tagged `version_2.4.0-beta2`).

lukasmatena, 2021-12-03:
@Pitel The easiest would be to checkout current master, build deps and do not rebuild them during bisection. The other option is to edit `deps/wxWidgets/wxWidgets.cmake` and plug in the latest hash (the 'actual' one, it should match the top of https://github.com/prusa3d/wxWidgets/commits/v3.1.4-patched).

Pitel, 2021-12-03:
But then, if it's a bug in some dependency, I won't hit it.

Anyway, if it's working correctly on `master` branch, I suppose it might get fixed somehow. So, I'll try using my build of Slicer for a weekend, and then let you know and hopefuly close the bug.

bubnikv, 2021-12-08:
Please don't forget to close this issue if it was resolved.

Pitel, 2021-12-08:
It seems fixed in beta3. 👍🏻

closed 2021-12-08 by Pitel