prusa3d/PrusaSlicer #7379: BUG: Unable to compile on macOS due to failure to compile TIFF

lbibass, 2021-11-29:
### Version
n/a

building on current master. 

Every time I try to compile the current master on macOS, it fails. 

```[6/59] Performing build step for 'dep_TIFF'
FAILED: dep_TIFF-prefix/src/dep_TIFF-stamp/dep_TIFF-build /Users/lbibass/git/PrusaSlicer/deps/build/dep_TIFF-prefix/src/dep_TIFF-stamp/dep_TIFF-build
cd /Users/lbibass/git/PrusaSlicer/deps/build/dep_TIFF-prefix/src/dep_TIFF-build && /opt/homebrew/Cellar/cmake/3.22.0/bin/cmake --build . --config Release -- -j10 && /opt/homebrew/Cellar/cmake/3.22.0/bin/cmake -E touch /Users/lbibass/git/PrusaSlicer/deps/build/dep_TIFF-prefix/src/dep_TIFF-stamp/dep_TIFF-build
[1/42] Linking C executable tools/fax2tiff
[2/42] Linking C executable contrib/dbs/tiff-palette
[3/42] Linking C executable contrib/dbs/tiff-rgb
[4/42] Linking CXX static library libtiff/libtiffxx.a
[5/42] Linking C executable tools/fax2ps
[6/42] Linking C executable contrib/dbs/tiff-grayscale
[7/42] Linking C executable contrib/dbs/tiff-bi
[8/42] Linking C executable tools/pal2rgb
[9/42] Linking C executable tools/ppm2tiff
[10/42] Linking C executable tools/thumbnail
[11/42] Linking C executable tools/tiff2ps
[12/42] Building C object contrib/iptcutil/CMakeFiles/iptcutil.dir/iptcutil.c.o
[13/42] Linking C executable tools/raw2tiff
[14/42] Linking C executable tools/tiff2bw
[15/42] Linking C executable tools/tiff2rgba
[16/42] Linking C executable tools/tiff2pdf
[17/42] Linking C executable tools/tiffcmp
[18/42] Linking C executable tools/rgb2ycbcr
[19/42] Linking C executable tools/tiffcp
[20/42] Linking C executable tools/tiffgt
FAILED: tools/tiffgt
: && /opt/homebrew/opt/ccache/libexec/cc -Werror=partial-availability -Werror=unguarded-availability -Werror=unguarded-availability-new -Wall -Winline -W -Wformat-security -Wpointer-arith -Wdisabled-optimization -Wno-unknown-pragmas -Wdeclaration-after-statement -fstrict-aliasing -O3 -DNDEBUG -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk -mmacosx-version-min=12.0 -Wl,-search_paths_first -Wl,-headerpad_max_install_names  tools/CMakeFiles/tiffgt.dir/tiffgt.c.o -o tools/tiffgt  libtiff/libtiff.a  port/libport.a  -lglut  -framework  OpenGL  -framework  OpenGL  /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk/usr/lib/libm.tbd  /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.0.sdk/usr/lib/libz.tbd  /Users/lbibass/git/PrusaSlicer/deps/build/destdir/usr/local/lib/libjpeg.a && :
ld: library not found for -lglut
```

I'm not sure why, it has compiled fine in the past.

spuder, 2021-12-09:
I'm also running into this and haven't found a solution. 

💡 Ideas
1. See https://stackoverflow.com/questions/16682156/ld-library-not-found-for-lgsl
2. Run `brew list | grep tiff` then `brew uninstall` any package that may be conflicting
3. Try `rm -r deps/build` then try and compile again.

lbibass, 2021-12-10:
I fixed it by completely removing all the command line tools and xcode, then reinstalling.

lbibass, 2022-01-13:
This is happening again. I removed all the tools and Xcode, but the problem didn't go away. I fixed it by adding -framework GLUT to deps/build/dep_TIFF-prefix/src/dep_TIFF-build/tools/CMakeFiles/tiffgt.dir/link.txt. 

I don't know why this doesn't get configured in the first place.

lbibass, 2022-01-13:
this goes away by updating to the newest TIFF, v4.3.0. I don't know why. 

Any chance that can be updated soon?

lbibass, 2022-02-04:
This is still happening. 

Please fix.

lukasmatena, 2024-02-19:
I assume this works now, after all the changes in the meantime. Closing.

closed 2024-02-19 by lukasmatena