prusa3d/PrusaSlicer #7381: [2.4.0 beta2] missing modifiers and other stuff

redPanther, 2021-11-29:
### Version
2.4.0 beta2 linux gtk3

### Operating system type + version
linux, ubuntu 18.04

### Behavior
when right click on imported object there is no "add part" "add modifier" and others in the menu.  further more in the right panel the List with height range modifiers are missing. Only Object manipulation is there.

version 2.4.0 beta2
[image]

version 2.3.3
[image]

did I miss something? is there a hidden new menu where all stuff is moved, or is here a bug?

greetings :-)

n8bot, 2021-11-30:
You likely have not yet enabled advanced or expert mode.

redPanther, 2021-11-30:
hmmm yeah how stupid. you are right. It is a bit interesting, because I have 2.3.3 2.4.0 alpha3 and 2.4.0 beta 2 in parallel on the system and for some reason on beta1 and beta2 it reseted my "expert" setting. 

Other weird thing, the "height range modifier" entry was available in the menu. I think this made no sense in "simple" mode.

bubnikv, 2021-12-04:
> Other weird thing, the "height range modifier" entry was available in the menu. I think this made no sense in "simple" mode.

Fixed with https://github.com/prusa3d/PrusaSlicer/commit/a8cc19127ebefb1762af458b8210ee94970165da

closed 2021-12-04 by bubnikv

n8bot, 2021-12-04:
> ... It is a bit interesting, because I have 2.3.3 2.4.0 alpha3 and 2.4.0 beta 2 in parallel on the system and for some reason on beta1 and beta2 it reseted my "expert" setting.

I forgot to mention, this is because PrusaSlicer will attempt to import settings it finds from previous versions, but only if it's of the same "type" or "better." As in, releases will only take configs from releases, betas will take configs from either betas or releases, and alphas will take configs from anything.

I'm not sure why id didn't ask to import your 2.3.3 settings, though... I think that is supposed to happen?