prusa3d/PrusaSlicer #7383: thin walls too long...

jmerc77, 2021-11-30:
### Version
Version 2.4.0-beta2+win64

### Operating system type + version
windows 10 21h2

### 3D printer brand / version + firmware version (if known)
cr10s marlin 2.x

### Behavior
thin walls producing lines that are too long.
enabling detect thin walls causes this.
should be shorter like this (from cura, imported into prusa gcode viewer):
[image]

should not look like this (but does):
[image]

project file:
[bolts.zip](https://github.com/prusa3d/PrusaSlicer/files/7623083/bolts.zip)

misc notes: not sure if it is a setting somewhere or an actual bug. it looks better with detect thin walls disabled and is not missing any walls.

dartrax, 2021-11-30:
Hi,
if you remove the raft and enable display of the shells, you'll see that the thin walls are not too long, they are just representing the thread of your model.

[image]

I suggest scaling the model or using a smaller nozzle for it, because the width of the threads is too narrow for a 0,4 nozzle. We may speculate why the thin wall is added only on top, not in the middle for every thread but the first one.

jmerc77, 2021-12-05:
is there some kind of threshold i can change? though, I really did not care if it matched the thread tips because it still physically works for my purposes. apparently cura is not applying thin walls (or at least not the same way)

dartrax, 2021-12-08:
I did not find any. May be someone else can explain why it skips the thin walls on all other threads but the first one.
If I were you I would disable thin wall detection if the output just works that way or I would adapt the CAD models (this one and the one of the matching counterpart) to make the threads so wide that two perimeters can fit in. A shape like a V instead of |_| could also help.

jmerc77, 2021-12-12:
this was my own design. but the size is desired. its for a remix. so other parts are not entirely mine.

jmerc77, 2021-12-14:
this issue is in beta4 as well incase anyone is curious.

closed 2026-09-29 by prusaslicer-bot[bot]