prusa3d/PrusaSlicer #7384: Проблемы с кодировкой

Chernysh-Aleksandr-Nikolaevich, 2021-11-30:
Версия 2.4.0-beta2 win10x64
В программе:
вкладка "Настройки принтера" >> вкладка "Заметки" >> текстовое поле:
Anycubic i3 Mega S
x              220мм
y              215мм
z              200мм
сопло          0,4мм
пруток         1,75мм
не правильно отображается кодировка для русского языка:
Anycubic i3 Mega S
x              220РјРј
y              215РјРј
z              200РјРј
СЃРѕРїР»Рѕ      0,4РјРј
РїСЂСѓС‚РѕРє    1,75РјРј

YuSanka, 2021-11-30:
Спасибо за ваш отчет, но мы не видем вашу проблемую. Нам бы очень помог скриншот.

Thanks for your report, but we can't reproduce your issue. Send us a screenshot. It can be very helpful for us.

Chernysh-Aleksandr-Nikolaevich, 2021-11-30:
[image]
[image]

bubnikv, 2021-12-04:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/8fc620375c1566d8b0c6becd4f6ed770c276d967

closed 2021-12-04 by bubnikv