prusa3d/PrusaSlicer #7385: No Sparse Layers (EXPERIMENTAL) Final Layer error

PompTel, 2021-11-30:
### Version
Version 2.4.0-beta2+win64

### Operating system type + version
Windows 11

### 3D printer brand / version + firmware version (if known)
Geeetech A20M Stock Model, no mods

### Behavior
This feature "No Sparse Layers" is brilliant, its saving me a lot of time and a lot of wasted filament.

However, the last layer of the print results in Print head returning the tower, and printing in the air.  Its not lowering to the last saved layer height of the print tower, which it does correctly throughout the entire print.  Its only the last layer

_Is this a new feature request?_

No its a noticed error on am experimental feature

#### Project File (.3MF) where problem occurs

Not required i dont think as its not relevant to any single multi extruder print

lukasmatena, 2021-11-30:
Sounds like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1355. Intended behaviour. There is no need to travel back to the tower with the very last nozzle purge, although it could be done. The "sparse" wipe tower does it all the time.

zzjlamb, 2021-11-30:
It is the same issue. I agree with Pomtel that this is a concern. I was hoping that it has been changed in this version. On a physical print this results in about 40cm or so of spaghetti being extruded over the tower, then dragged across to the back left corner.
I agree that ramming is very important to avoid mishapen tips, but is there any reason that this shouldn't be extruded on to the purge tower? It is a very messy way to end a print.
Also wouldn't it also be better to do cooling moves AND ramming to make a good tip for the next print?

bear454, 2022-10-12:
I'm so glad I found this was already reported as an issue... it's been driving me bonkers. Here's a photo of the slice showing the extrusion-in-space:
[image]

closed 2026-09-29 by prusaslicer-bot[bot]