prusa3d/PrusaSlicer #7386: Unable to compile on Linux since last friday

plampix, 2021-11-30:
### Version
Current master

### Operating system type + version
Debian GNU/Linux 11

### 3D printer brand / version + firmware version (if known)
Not relevant

### Behavior
/home/peterp/Software/PrusaSlicer/src/slic3r/Utils/OctoPrint.cpp: In member function ‘virtual bool Slic3r::OctoPrint::upload(Slic3r::PrintHostUpload, Slic3r::PrintHost::ProgressFn, Slic3r::PrintHost::ErrorFn) const’:
/home/peterp/Software/PrusaSlicer/src/slic3r/Utils/OctoPrint.cpp:127:44: error: ‘boost::nowide’ has not been declared
  127 |         std::string resolved_addr = boost::nowide::narrow(test_msg);
      |                                            ^~~~~~

bubnikv, 2021-11-30:
Hopefully fixed with c514f4526066be6279d33937a697e68a80081e63. Please let
me know whether that helped.

plampix, 2021-11-30:
It did, thank you!

closed 2021-11-30 by plampix

P-C-R, 2021-11-30:
same here! I used and old octoprint.cpp yesterday to compile. but now it works ;)