prusa3d/PrusaSlicer #7387: [FR] Readme on how to build the AppImage

P-C-R, 2021-11-30:
Lately i switched completly to linux for all my CAD work. 
I have no problem to compile PS and start ./prusa-slicer and add this to the Desktop. Thank you for the option in the menu bar!
It would be really nice to see a Readme on how to build the AppImage? With that i could integrate it a little bit better.

Does someone has a readme for it or would like to write one?

Best

Area5142, 2021-11-30:
When I build an AppImage from the compiled PrusaSlicer binary, I use the OS tool "appimagetool-x86_64.AppImage" and an existing working AppImage.

- Extract files from existent working PS AppImage (from Prusa GitHub).
- Update files in unpacked app image with bin and resources from build.
- Repack Appimage.

I know that this is not a build from scratch, but it has been working for me since PS version 2.3.

P-C-R, 2021-12-01:
@Area5142 that worked!
I included all resources folders and the both Prusa-slicer and gcodeviewer files. Did i miss something?

Area5142, 2021-12-01:
@P-C-R I replace files in the working app image with the newly build version (resource+binary) and repack.
To get the gcode viewer to work it is enough to make a sym-link to the new app image and name it "prusa-gcodeviewer". This will make PS start as gcode viewer.

closed 2026-09-29 by prusaslicer-bot[bot]