prusa3d/PrusaSlicer #7389: Unable to upload gcode to Octoprint from 2.4.0-beta2-win64

ochm, 2021-11-30:
### Version
2.4.0-beta2-win64

### Operating system type + version
Windows10

### Behavior
When i try to upload gcode, i see error message window:
Error uploading to print host:
HTTP 503: <html><body><h1>503 Service Unavailable</h1>
No server is available to handle this request.
</body></html>

When i use beta1 or any previous version of slicer, upload is well.

I'm using Octoprint for two printers, the url in physical printer definition is
for printer1 http://193.85.73.245/P1/
for printer2 http://193.85.73.245/P2/

Project: [zasobnik.zip](https://github.com/prusa3d/PrusaSlicer/files/7627301/zasobnik.zip)

neophyl, 2021-11-30:
Could it be the same as this https://github.com/prusa3d/PrusaSlicer/issues/7361 ?

ochm, 2021-12-01:
> Could it be the same as this #7361 ?

I don't know, I'm not using hostname in definition, I'm using IP, and i don't know nothing about port 5000, 
HAproxy in Octoprint is listening on 80 and 443
# netstat -nap|grep hap
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      638/haproxy
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      638/haproxy

The definition of connection in physical printer setting does not contain any port,
it is written as standard HTTP: http://ip.add.re.ss/P1/ so it means port 80.

bubnikv, 2021-12-01:
> Could it be the same as this #7361 ?

It surely is.

ochm, 2021-12-01:
> > Could it be the same as this #7361 ?
> 
> It surely is.

I'm not so sure, because I'm not using nonstandard port in physical printer definition (using http), and my error message is number 56 (who knows what does it mean).
[image]

But if you are sure, my issue can be closed.

lukasmatena, 2021-12-01:
@ochm Can you please try slicing exactly the same 3MF in beta1 and see if it works there? If it does not work, it is likely on your end. If it does, we may provide you with a dev build where #7361 is fixed, so we can make sure it was indeed the problem. Thanks.

kocikdav, 2021-12-01:
It is a different manifestation of same bug I introduced a week ago.  But thanks to this issue I understood better how to fix it. I will paste a dev build here shortly.

ochm, 2021-12-01:
> It is a different manifestation of same bug I introduced a week ago. But thanks to this issue I understood better how to fix it. I will paste a dev build here shortly.

Great, thank you!

kocikdav, 2021-12-01:
Please try this [build](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/DK/PrusaSlicer-2.4.0-beta2+43-dk-printhost-win64-ge398289d3-202112010937.zip). (if prompted before download use login prusaslicer : slicer).  Also please note that this build is experimental and other features could be broken or in middle of development.

When you run PrusaSlicer with command line, there should be output (marked as error but its not an error) to log real address for upload.

ochm, 2021-12-01:
Yes! Works OK, like in beta1

closed 2021-12-01 by FidelCapo