prusa3d/PrusaSlicer #7391: [BUG] 0.6mm nozzle + height range modifier + layer height value = (overly) constraint error

Upcycle-Electronics, 2021-11-30:
### Version
Version 2.3.3 (flatpak)

### Operating system type + version
(lazy copied from FreeCAD's version report output)
OS: Fedora 34.20211016.0 (Silverblue) (GNOME/gnome)
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)

### 3D printer brand / version + firmware version (if known)
user is part of the MK3S+kitty army

### Behavior
* when slicing using a 0.6mm nozzle layer height modifier, the layer height setting is messed up. The text in the "layer height" setting must be between the values that are already setup in the system presets (0.15mm-0.40mm). For example, it is not possible to set the height range modifier to 0.1mm.

WORK AROUND
* The problem is only in the height range modifier setting, so a possible user work around is to set the entire model to slice at 0.1mm layer height, then use a height modifier to set a value to one that is within the system preset profiles

* the problem came up when I was trying to slice a model with a thread at the top of the part and needed a 0.1mm layer height to prevent dropping overhangs on the thread

Thanks for taking the time and your efforts.
-Jake

FidelCapo, 2021-12-01:
Can you please attach the 3mf file? Thank you.

Upcycle-Electronics, 2021-12-02:
[Try2ChangeLayerHeightModifierFrom-p15To-p10.zip](https://github.com/prusa3d/PrusaSlicer/files/7638250/Try2ChangeLayerHeightModifierFrom-p15To-p10.zip)
No problem.

bubnikv, 2021-12-10:
Printer profile, Extruder 1 minimum layer height is set to 0.15, this is enforced.
If you want to print lower layer heights, you should lower the value there.

closed 2021-12-10 by bubnikv

Upcycle-Electronics, 2021-12-10:
The main overall layer height for the profile can still be set to lower than the extruder 1 minimum layer height as was the case in the work around I mentioned. Use the same file I included earlier and change Layers and perimeters > Layer height to 0.1mm. This works. The issue is not resolved, it is still an error.

bubnikv, 2021-12-11:
I see the inconsistency. The inconsistency is in your profiles, however PrusaSlicer does not handle it gracefully.

What do your recommend then? Should we stop slicing such an invalid profile (layer height lower than the minimum layer height allowed)?

Upcycle-Electronics, 2021-12-12:
I make no claim about knowing how to do your job or what the best design may be. In my case, I assumed this kind of setting would not be something I need to troubleshoot on a MK3S+. Is there some reason these default values need to be constrained to anything a user might intentionally enter?

bubnikv, 2021-12-13:
> Is there some reason these default values need to be constrained to anything a user might intentionally enter?

Very thin or very thick layers are not printable. Thin layers are not printable due to plastic viscosity issues for example.

Upcycle-Electronics, 2021-12-17:
lol
very committed you are
Are you saying 0.1mm and 0.45mm layer heights are not printable with 0.6mm nozzles in general on any machine, or is this a statement of limitation on the Prusa MK3s+?

bubnikv, 2021-12-20:
> very committed you are

Lol, I think I am committed more than you would think.

> Are you saying 0.1mm and 0.45mm layer heights are not printable with 0.6mm nozzles in general on any machine, or is this a statement of limitation on the Prusa MK3s+?

It is really a question on @rtyr why he set the minimum layer height of 0.15mm for our MK3S printer profiles for 0.6mm nozzle.