prusa3d/PrusaSlicer #7394: FEATURE REQUEST STAMP MODE

TheGrimReefer, 2021-12-01:
Any chance we could get something 
 similar to the stamp feature in Canvas?

FidelCapo, 2021-12-01:
Can you please add more information? Some examples? Thank you

TheGrimReefer, 2021-12-01:
put a link at the bottom with more in depth info. 

Stamping; a tool that allows CANVAS users to take 2D images and project them onto a 3D model with just a few clicks. 

When you upload an image using the Stamp tool, CANVAS analyzes the image and creates groups of similar colors. Once these have been created, CANVAS uses the colors from your project and matches them to the groups of colors from the image. These can be changed to any color or it can be given a transparent background. Once all the color groups have been assigned to one of the four drive colors, the image is ready to be stamped onto the model.

https://support.mosaicmfg.com/Wiki/Stamping_Tips

rtyr, 2021-12-01:
There is already FR for this feature.

Closing as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6916.

closed 2021-12-01 by rtyr