prusa3d/PrusaSlicer #7397: Slicer generate strange trajectories for my model

projectiondp, 2021-12-01:
### Version
_Version 2.3.3+win64_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Slicer generate strange trajectories for my model_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_No_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)
[CLAW_KNOB.zip](https://github.com/prusa3d/PrusaSlicer/files/7635807/CLAW_KNOB.zip)
_

n8bot, 2021-12-01:
It is unclear exactly what you're talking about, but let me take some guesses:

There appear to be cracks or gaps on the wall of the ID. These appear to be regions which are too thin to produce extrusion and the slicer is not generating toolpaths there. Or, perhaps, the model is simply not ready for 3D printing, with overlapping shells, small holes, etc.

Other than that, the only thing I see that might raise questions are the thick single extrusions on the top, which is proper and intended but can be disabled with the setting "detect thin walls."

There doesn't seem to be anything in this picture that shows a problem with the slicer software.

projectiondp, 2021-12-01:
Thanks a lot!

You are right! This is a problem of settings. After disabling "detect thin
walls" the trajectories are amazing! Thank you so much!!!

rtyr, 2021-12-02:
Resolved. Closing.

closed 2021-12-02 by rtyr