prusa3d/PrusaSlicer #7398: Can not compile latest source (added lightning infill)

P-C-R, 2021-12-02:
`[ 30%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/Format/AMF.cpp.o
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:420: src/libslic3r/CMakeFiles/libslic3r.dir/Fill/Lightning/DistanceField.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/pcr/AppImages/PrusaSlicer/src/libslic3r/Fill/Lightning/TreeNode.cpp: In function ‘bool Slic3r::FillLightning::inside(const Polygons&, Slic3r::Point)’:
/home/pcr/AppImages/PrusaSlicer/src/libslic3r/Fill/Lightning/TreeNode.cpp:148:53: error: ‘PointInPolygon’ is not a member of ‘Slic3r::ClipperLib’
  148 |         const int is_inside_this_poly = ClipperLib::PointInPolygon(p, poly.points);
      |                                                     ^~~~~~~~~~~~~~
/home/pcr/AppImages/PrusaSlicer/src/libslic3r/Fill/FillLightning.cpp: In member function ‘virtual Slic3r::Polylines Slic3r::FillLightning::Filler::fill_surface(const Slic3r::Surface*, const Slic3r::FillParams&)’:
/home/pcr/AppImages/PrusaSlicer/src/libslic3r/Fill/FillLightning.cpp:16:52: error: invalid use of incomplete type ‘const class Slic3r::Surface’
   16 |     return layer.convertToLines(to_polygons(surface->expolygon), generator->infilll_extrusion_width());
      |                                                    ^~
In file included from /home/pcr/AppImages/PrusaSlicer/src/libslic3r/Fill/FillLightning.hpp:4,
                 from /home/pcr/AppImages/PrusaSlicer/src/libslic3r/Fill/FillLightning.cpp:3:
/home/pcr/AppImages/PrusaSlicer/src/libslic3r/Fill/FillBase.hpp:20:7: note: forward declaration of ‘class Slic3r::Surface’
   20 | class Surface;
      |       ^~~~~~~
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:459: src/libslic3r/CMakeFiles/libslic3r.dir/Fill/Lightning/TreeNode.cpp.o] Error 1
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:407: src/libslic3r/CMakeFiles/libslic3r.dir/Fill/FillLightning.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1215: src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make: *** [Makefile:160: all] Error 2`

hejllukas, 2021-12-02:
It should be fixed with e85a0ba248829c8d0dcc4f6aec6a39685f451cc0. Please let me know if it helped.

P-C-R, 2021-12-02:
will do when i am on my machine again

P-C-R, 2021-12-02:
sadly not

`[ 35%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/GCodeWriter.cpp.o
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:316: src/libslic3r/CMakeFiles/libslic3r.dir/Fill/FillAdaptive.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:1215: src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
`

hejllukas, 2021-12-02:
I think that compilation was terminated because of not enough memory during the compilation.
You can try to limit the number of threads during the compilation using `-j N`, where `N` is the maximum number of used threads. For example `make -j 2`. If you are already using it, try to decrease its value.

P-C-R, 2021-12-03:
@hejllukas  you were right ;)

closed 2021-12-03 by P-C-R