prusa3d/PrusaSlicer #7399: PrusaSlicer crashes when the wipetower is moved by dragging on build plate

ssill2, 2021-12-02:
### Version
_Version of PrusaSlicer used goes here_
2.4.0 built from master as of Dec 1, 2021
9ea81a9dbfcde165b5dab56bf4c8c1f07825c4d1

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
 windows 11 64bit

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
3
MK3S+ with MMU2S

### Behavior
* _Describe the problem_
PrusaSlicer crashes when I open a particular .3mf and simply move the wipetower around by dragging it.

* _Steps needed to reproduce the problem_
Open .3mf and move the wipe tower a few times.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[SquidGame_luggage_tag.zip](https://github.com/prusa3d/PrusaSlicer/files/7639139/SquidGame_luggage_tag.zip)

lukasmatena, 2021-12-02:
Thanks. Reproduced in alpha2 and later, it does not crash in alpha1. `std::vector::back` is being called on an empty vector in GCodeViewer.cpp:1193 (`if (prev.type != curr.type || !buffer.paths.back().matches(curr)) {`). Probably something for @enricoturri1966 .

lukasmatena, 2021-12-02:
Previous comment was supposed to go to a different issue. As for this one, I did not manage to reproduce it in beta2 nor current master (on Windows).

bubnikv, 2021-12-02:
I was not able to reproduce the crash either. Are you still able to
reproduce the crash with the current master? What exactly did you do? Do
you have background processing enabled?

rtyr, 2021-12-02:
I was not able to reproduce it either on current master. With or without background processing.

ssill2, 2021-12-02:
I just left for a few days of vacation but I will come back and pull
latest.  I was also messing multi part objects too.  It’s happened several
times.  But I will follow up next week

ssill2, 2021-12-12:
I just got back from being away.  currently doing a complete clean and build to test this again.

ssill2, 2021-12-12:
alright, doing a complete clean/build of deps+ps made this issue go away thanks!

closed 2021-12-12 by ssill2