prusa3d/PrusaSlicer #7404: Colorchanges with single Extruder @Option "sequential printing" with multiple Objects not possible

funtomaas, 2021-12-03:
### Version
Version: 2.3.3+win64
Build: PrusaSlicer-2.3.3+win64-202107161027

### Operating system type + version

Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19043), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa Mini+

### Behavior
* _Describe the problem_
addition in layers (for Color-/ Filament-changes) not possible (no "+" sign in right bar in "Preview"-Screen), if printing multiple Objects with printing-Option: "sequential-printing" activated.

* _Steps needed to reproduce the problem_
see above
* _Expected Results_
addition of Filamentchange possible
* _Actual Results_
*not possible (no "+" sign etc.)

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

_Is this a new feature request?_
no

bubnikv, 2021-12-08:
PrusaSlicer 2.4.0 shows a tooltip explaining, that color change is not supported in sequential mode.

funtomaas, 2021-12-09:
OKay, thanks a lot!

Hologos, 2022-02-03:
> PrusaSlicer 2.4.0 shows a tooltip explaining, that color change is not supported in sequential mode.

@bubnikv: Is it a temporary thing? I can see no reason why it couldn't be supported. Thank you

scottamain, 2022-04-19:
I'd also love to see this and am surprised to see the tooltip claim it is "impossible." It seems totally possible, although I don't know what the technical challenges might be.

3DCoded, 2022-07-27:
+1 Please implement this! I have many projects and are just waiting to click the print button when this gets fixed!

plug-it-in, 2022-08-04:
Up vote from me as well

Hologos, 2022-08-08:
In the meantime, I manually add M600 into the generated gcode file. Would be great if it was supported.

crabbedh, 2022-08-13:
Absolutely no reason *not* to implement this. It allows one to create multiple parts in different colors for visual testing without having to manually remove/reload each part. Would be a very welcome addition.

Hologos, 2022-08-13:
> Absolutely no reason *not* to implement this. It allows one to create multiple parts in different colors for visual testing without having to manually remove/reload each part. Would be a very welcome addition.

You can add M600 (or the sequence you want) to "on next object" gcode.

crabbedh, 2022-08-13:
> > Absolutely no reason _not_ to implement this. It allows one to create multiple parts in different colors for visual testing without having to manually remove/reload each part. Would be a very welcome addition.
> 
> You can add M600 (or the sequence you want) to "on next object" gcode.

From my understanding of PrusaSlicer 2.4x (& 2.5) you'd do this by pausing at layer height in preview window, but this is disabled for sequential printing. So the only option I can think of (but am not that experienced with Slicer) is to generate the GCODE file and then manually insert it at the right place. Right now it is faster for me to print individual models. Don't think there is an easier way to do this.. maybe Octoprint supports it easier.. I can look.

Hologos, 2022-08-13:
@crabbedh You dont have to do it like that. Use the gcode box "on object change".

crabbedh, 2022-08-13:
Maybe not vague for others.. (so explain it to me if another understands).. but I'm not sure where/what the "gcode box" is. Is this a box that appears in PrusaSlicer if I click in the right places?  If so, where? Is this a box in OctoPrint? Or maybe point me to a post somewhere explaining it. I searched on Prusa's forums for your key words.. and nothing appeared that would describe what you refer to.

Hologos, 2022-08-13:
> Maybe not vague for others.. (so explain it to me if another understands).. but I'm not sure where/what the "gcode box" is. Is this a box that appears in PrusaSlicer if I click in the right places? [...]

Printer Settings > Custom G-code > Between objects G-code (for sequential print)

Save is ideally as a separate config so you can set it only when you want. If you want to sequentially print with the same color, it would stop after every object and asked you to change the filament.

[image]

crabbedh, 2022-08-13:
ahhh... super.. thx so much for the explanation.. that would do what I want..

Vashiru, 2022-08-29:
I suppose there are 2 ways of doing this (just sharing my thoughts for people to comment on, don't have the solution):

The 'swap-colors-for-each-object' way and the 'swap-colors-only-for-each-color' way. What I mean by this is the following:

Say you have a simple 2 color keychain of which you want to print 9 copies, first a few layers in black (say 10), followed by some in white (say 5). The first one would go:

**Swap-colors-for-each-object:**
1. Start object 1 in black
2. Swap color to white, finish object 1
3. Swap color to black, start object 2
4. Swap color to white, finish object 2
5. Continue until object 9.

If you have 9 objects to print, that means you have to go through 18 filament load/unloads (not to mention probably cleaning).

The alternative however would be:
**Swap-colors-only-for-each-color:**
1. Start object 1 in black up to layer 10
2. Followed by the black section of object 2
3. Followed by the black section of object 3
6. Continue this until the black section of object 9 has been completed
7. Swap color to white
8. Finish object 1 in white
9. Finish object 2 in white
10. Continue this until object 9 in white is finished

I understand that the latter method would only work for objects that actually have the color change at the same height for each object (or close to the same height, but that would make things way more complicated I'd figure). But it would definitly make mass printing certain items a lot easier.

kubispe1, 2022-09-03:
#3944

loadedsith, 2024-02-29:
Hi y'all, I wanted to point out a use case for this feature: [HueForge](https://shop.thehueforge.com/)  Transmission Distance with the Step Test prints. [Here's a short video about these prints](https://www.youtube.com/watch?v=39uGtPLUr88).

It would be great to fill the bed with these test prints, then have 2 color changes repeated for each item: one at the start to lay down the base (black or white) layers, and another at the 12th layer to swap to the test color. This would automate the process of creating many of these little test pieces, one for each filament color.

Thanks.

Isaac-dot-py, 2026-09-21:
This is even more of an issue in prusaslicer 3.0, where it doesn't even display a helpful tooltip and instead just silently deletes the color change with no explanation.

Jan-Soustruznik, 2026-10-02:
Hello, thank you for your feedback. I noticed there is already a similar feature request. To keep everything organized in a single thread, we are consolidating this topic into discussion (#15961). Therefore, I am closing this issue as duplicate of #3944 . Have a great day.

closed 2026-10-02 by Jan-Soustruznik