prusa3d/PrusaSlicer #7405: Size manipulation of modifiers producing unexpected scaling

edwardhorsford, 2021-12-03:
### Version
2.3.3

### Operating system type + version
MacOS 11.6

### 3D printer brand / version + firmware version (if known)
Prusa Mini+

### Behavior
I'm adding a modifier to a block, and want to set the exact dimentions.

In the video below, I attempt to type in a dimension of 25.4mm - when I leave the field, this changes to 42.33mm
Similarly changing other scale values produces unexpected values.

### What I expected

I expected that setting width of 25.4mm would set the width to 25.4mm - instead it's a much larger value.

Note my file is just primitives drawn directly in PrusaSlicer 

#### Project File (.3MF) where problem occurs
[image]
[Shape-Cylinder.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7650716/Shape-Cylinder.3mf.zip)

neophyl, 2021-12-04:
I was just trying the latest beta (2.4b3) and this seems to work fine in that.  So I think this has already been fixed.

lukasmatena, 2021-12-06:
This was fixed between 2.4.0-alpha3 and beta1. From the change log:
> Fixed non-uniform scaling of a volume using the part manipulation panel #6380.

Closing.

closed 2021-12-06 by lukasmatena