prusa3d/PrusaSlicer #7406: Supports Aren't Generated Under Enclosed Horizontal Surfaces

kimkash, 2021-12-03:
### Version
2.3.1

### Operating system type + version
Win10 Pro, 21H2 OS Build 19044.1348

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa MK3S+,stock, firmware 3.10.0-4481

### Behavior
Problem: Supports aren't generated under horizontal surfaces when said surfaces are surrounded by walls touching the build plate.
Tried to add support enforcer with PrusaSlicer built-in feature or by creating another body in Solidworks and importing as multibody 3MF and assigning second body as support enforcer same result.
Added a slight slope (89 degrees) to see if there's change, now there's support but only around 70% of the surface and it vanishes near the top part of the slope.
Tried the same method described for the horizontal surface, same results.
Important to note: in 'put supports everywhere' mode, still doesn't put supports properly, not just 'only build plate'/'only support enforcers'

Steps to reproduce:
I've added files and screenshots to make it easier, but here some short instructions:
*create a simple box, meaning a cube with rectangular hole in one of it faces.
*Place it on the build plate so the holed face pointing down.
*Try to create supports

Expected Results:
Supports below the horizontal/89Deg surface
Actual Results:
for completely horizontal (90Deg)- no supports at all, for 89Deg- supports about 70% of surface.

Files:
[image]
[image]
[image]
[image]
[image]
[PrusaSlicer Bug-3MF Files.zip](https://github.com/prusa3d/PrusaSlicer/files/7651143/PrusaSlicer.Bug-3MF.Files.zip)

n8bot, 2021-12-03:
I think it is because you have the setting "don't support bridges" enabled.

I'm basically 100% sure it is that. 😁

bubnikv, 2021-12-07:
> I think it is because you have the setting "don't support bridges" enabled.

I think the same.
Closing

closed 2021-12-07 by bubnikv