prusa3d/PrusaSlicer #7407: "File not found" when saving to "Documents" folder in Windows

gordo3di, 2021-12-03:
### Version
_4.2 Beta 2_

### Operating system type + version
_Windows 10 64-bit_

### 3D printer brand / version + firmware version (if known)
_Irrelevant_

### Behavior
Whenever I try to do a "Save-as" of the .3mf file to "Documents" in Windows (ie. C:\users\username\documents\...") it comes up as "File not found".

Not a big issue but I believe I had the same issue in Beta 1 as well. I do not believe I saw it in 2.3.3

bubnikv, 2021-12-04:
Is there anything special about your file system with the documents folder? Is it encrypted, on network? Any antivirus or 3rd party security system? Any sand boxing of PrusaSlicer is active?

bubnikv, 2021-12-06:
@gordo3di 

Is it a company laptop? Or some Windows 10S installation?
Did your admin put some restrictions in place?

gordo3di, 2021-12-06:
@bubnikv 

Nope just a normal installation with standard admin restrictions. Not windows 10S.

I just tried from my home computer and it seems to save fine. Weird, something must just be up with my windows install. Thanks for the help. I'll close this as it must just be a local issue.

closed 2021-12-06 by gordo3di

bubnikv, 2021-12-06:
Didn't you experiment with sandboxing? I did not try that, but there is
supposed to be such a feature in Windows 10 professional or enterprise.

gordo3di, 2021-12-06:
I haven't experimented with it. Apparently I can't turn it on because my processor is too old haha.