prusa3d/PrusaSlicer #7408: Crash after closing error for deleting box shape with cylinder negative volume

jtbalk, 2021-12-03:
### Version
PrusaSlicer-2.4.0-beta2+linux-x64-202111261557.AppImage

### Operating system type + version
MX-19.4_KDE_x64 patito feo November 11 2020

### Behavior
Deleting a box shape which has a cylinder negative volume attached causes an error. After the error is closed, PrusaSlicer crashes.
To reproduce:
* right click build plate> add shape> box
* right click box > add negative volume > cylinder
* click on the box in the right side object list to select ONLY the box (deselect the negative volume cylinder)
* trying to delete only the box will produce error (From Object List You can't delete the last solid part from object)
* click OK or close on the error message > crashes here

Interestingly, the crash doesn't seem to happen if you use, for example, a box shape with a box negative volume - the error will pop up as expected, but PrusaSlicer will not crash. Sorry, I did not test more combinations of boxes and cylinders and modfiers, but box shape with cylinder negative volume will reproduce a crash.

n8bot, 2021-12-03:
I forgot to report this, I experienced it to. It seems to happen any time you try to delete the parent model before deleting the children.

bubnikv, 2021-12-04:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/8a2c15ad606a42732d992fea66ae2fb8a483d119
Closing

closed 2021-12-04 by bubnikv