prusa3d/PrusaSlicer #7412: [Question] What CI system builds OSX Packages?

spuder, 2021-12-04:
I'd like to make a change to the OSX Installer package, however I don't see any pipelines to build these packages automatically. (.github/actions, .travis-ci, .circleci). The only documentation I can find is how to compile manually

https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md

How can I create a pull request to modify the OSX build process?

n8bot, 2021-12-05:
What change do you want to make?

I don't know how Prusa team does it, but [SuperSlicer fork has build instructions via the actions page](https://github.com/supermerill/SuperSlicer/actions/runs/1487929143/workflow), which shows how he makes the DMG. It works for PrusaSlicer too, if you skip the irrelevant parts.

spuder, 2021-12-05:
Thank you. I also found a build script on that fork which is helpful since I'm having a hard time compiling by hand. 

- [BuildMacOS.sh](https://github.com/supermerill/SuperSlicer/blob/master/BuildMacOS.sh)
- [BuildLinux.sh](https://github.com/supermerill/SuperSlicer/blob/master/BuildLinux.sh)
 
 I can open a separate issue, but I'd like to add back the 'Drag to Install' folder shortcut that used to exist in old versions of the DMG (Back when it was bundled in prusa drivers)
 
 Current
[image]

Origional
[image]

n8bot, 2021-12-05:
You can just drag that icon to the applications folder. Doesn't that work? Or do you want the fancy graphics? Or do you want the drivers also?

spuder, 2021-12-05:
It's a minor inconvenience but it takes the user 3 clicks to go open a separate finder window and navigate to the Applications folder. Additionally users new to OSX may not be aware how to do that. 

Im also interested in helping the PrusSslicer project, but with limited C++ experience there is only so much I can do. Adding cosmetic improvements to the installer seems like something within my skill set.

n8bot, 2021-12-05:
Very cool! Well, I don't know at all how to do that but I wish you luck that sounds fun.

lukasmatena, 2021-12-06:
> I can open a separate issue, but I'd like to add back the 'Drag to Install' folder shortcut that used to exist in old versions of the DMG (Back when it was bundled in prusa drivers)

lukasmatena, 2021-12-06:
> I can open a separate issue, but I'd like to add back the 'Drag to Install' folder shortcut that used to exist in old versions of the DMG (Back when it was bundled in prusa drivers)

I have a feeling you are trying to solve a non-existing issue. After 2.4 final is released, the Drivers bundles will be updated and they will use the same installer as before. I don't know if github DMGs (especially the prerelease ones) need that kind of attention. For example, Windows installer is also only included in the Drivers package.

spuder, 2021-12-06:
Thank you for the clarification. You are right, perhaps the direct downloads of prusa slicer isn't the preferred install method. Most users would still be using the Drivers Bundle which does have the shortcut. 

I'd still like to make contributions to prusa slicer where I can. The build process appears to be manual and there isn't any documentation about it. I'd like to help improve/automate building prusa slicer, but need to know where there is need.

lukasmatena, 2021-12-07:
@spuder The build process is documented in this [wiki page](https://github.com/prusa3d/PrusaSlicer/wiki/Building-PrusaSlicer) (which in fact just points to `doc` folder in the tree). We are trying to keep the documentation updated. The build is quite simple when one follows the guide and does not try to use system dependencies (which is the main source of build problems reported here).

I personally don't think that anything needs automation and although the idea is brought up every now and then, I frankly don't see the benefit. And we at PR already have a build server to take care of. We are using our custom internal build server, which is not reachable from the outside. It uses [buildbot](https://buildbot.net/) and couple of custom build scripts. It works well, builds fast and does all that we need it to do. Its purpose is to continually provide testing builds from 8 developers to the testers, but that is something that an external contributor should not need.

spuder, 2021-12-07:
Thats what I wanted to know, thank you. I'm glad to hear that you have a system that works well. 
From my point of view, I had a lot of problems trying to compile on my mac. It would have been nice to have a CI system that could build my branch for me to avoid these dependency problems. I feel the barrier-to-entry is a bit daunting for new contributors to the prusa slicer project. 

From your point of view, you are right, it probably isn't worth the disruption/cost to overhaul the build process to make it accessible to everyone. 

Thanks for the feedback. I'll keep looking for other areas where prusa slicer could be improved that aligns with my skillset (Or I'll go learn c++)

closed 2021-12-07 by spuder

lukasmatena, 2021-12-07:
>  I had a lot of problems trying to compile on my mac

What exactly did you run into? If you sticked to [the macOS build guide](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Mac%20OS.md) and still had problems, we would be thankful if you reported or even fixed them. From my point of view, documenting the build process well is the way to go. It might lower the entry barrier enough.

To be honest, I can hardly imagine doing anything non-trivial without being able to compile PrusaSlicer locally, not to mention debugging, asserts and similar. But I admit I have by far not seen everything.

spuder, 2021-12-09:
Here are the issues I've run into trying to compile on OSX. (I never got it working, luckily I have a second mac I _did_ get it to compile on)

Compile Issues

- https://github.com/prusa3d/PrusaSlicer/issues/3416
- https://github.com/prusa3d/PrusaSlicer/issues/2624#issuecomment-569446331
- https://github.com/prusa3d/PrusaSlicer/issues/7379

I've researched a few possible solutions to provide more portable/repeatable builds on workstations. Here is what I've found. 

1. [Vagrant + OSX virtual machine](https://app.vagrantup.com/AndrewDryga/boxes/vagrant-box-osx)
  `Pros`: very likely would work, would work for linux/windows users too
  `Cons`:  would require lots of disk space / ram, experimental, 

2. [Run docker on OSX](https://github.com/sickcodes/Docker-OSX)
  `Pros`: Docker is popular tool
  `Cons`: experimental, would work for linux too (possibly windows users)
  
 3. Nix
   `Pros`: Lighter weight isolation than docker or virtualmachine, becoming very popular with linux developers
   `Cons`: New, fairly steep learning curve, linux/mac only (no windows)
   
  4. OSX Cloud
    `Pros`: Simplicity
    `Cons`: Cost, likely overkill for development, no linux/windows
    
    5. [XCode Symroot](https://developer.apple.com/library/archive/documentation/DeveloperTools/Reference/XcodeBuildSettingRef/0-Introduction/introduction.html#//apple_ref/doc/uid/TP40003931-CH3-SW39)
      `Pros`: Supported by Apple
      `Cons`: I've never used it , no linux/windows

bubnikv, 2021-12-09:
>    - #2624 (comment)

I am not sure whether we can help you. None of us PrusaSlicer developers
are particular Apple fans (just our tester Filip is), we develop mostly on
Windows or Linux and we resort to solving OSX issues only when necessary.
We don't touch our development OSX machines as long as they work.

First, if one of your machines is ARM, then the last time I had to install
the development system on that machine, brew was not quite supported, and
it took a while and a couple of workarounds that I did not document to make
it work.

Second, we only support static compilation of PrusaSlicer, that is with
most dependencies compiled by our deps script linked statically. If you
installed some packages (libtiff, wxwidgets...) from brew, then this is an
untested situation and the cmake system may break. Sorry, we don't have
time to troubleshoot such configurations.

lukasmatena, 2021-12-09:
@spuder 
> Second, we only support static compilation of PrusaSlicer, that is with most dependencies compiled by our deps script linked statically

I'd like to stress this part. I helped several people here with "wxWidgets not found" on Linux and in most (if not all) cases it is caused by trying to link to system libraries, which are rejected as too old. This is not a supported scenario, in the sense that we will not solve the dependency hell for anyone doing that. 

People usually don't bother to report their system version or share their cmake invocation, but instead report "wxWidgets not found", "Fix the bug" or "ran into this as well". Then it gives the impression that the cmake scripts are utterly broken. I'm not saying there can't be a problem, but in most cases reported here there wasn't.