prusa3d/PrusaSlicer #7413: Variable layer heigth force-close in Linix x64

xitix, 2021-12-04:
### Version
_Version of PrusaSlicer used goes here_
Version 2.4.0-beta3+linux-x64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
when click on variable layer heigth icon, PrusaSlicer force-close.
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* remediate the issue
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
https://youtu.be/FGZ-CKqcT7w

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2021-12-04:
No issue on Windows, no issue on WSL2 (virtualized Linux) with software OpenGL.

Anybody else has the same issue?

hejllukas, 2021-12-04:
I tried to reproduce it on native Ubuntu 20.04 and also on Ubuntu 20.04 in VirtualBox with WindowMaker, and there is no issue either.

What distribution and its version are you using? Could you please also provide us the content of menu _Help -> System Info dialog_?

xitix, 2021-12-05:
> I tried to reproduce it on native Ubuntu 20.04 and also on Ubuntu 20.04 in VirtualBox with WindowMaker, and there is no issue either.
> 
> What distribution and its version are you using? Could you please also provide us the content of menu _Help -> System Info dialog_?

2.4.0 beta3 it behaves ok on Windows 10 in VirtualBox.

The  force-close issues appear with all packages  on Linux x64, so far have experienced same issue with:
1. PrusaSlicer-2.4.0-beta3+linux-x64-202112040926.AppImage
2. PrusaSlicer-2.4.0-beta3+linux-x64-202112040926.tar.bz2
3. PrusaSlicer-2.4.0-beta3+linux-x64-GTK3-202112040936.AppImage

The issue:
PrusaSlicer-2.4.0-beta3+linux-x64-./prusa-slicer
[2021-12-05 10:18:48.035520] [0x00007f3d873698c0] [trace]   Initializing StaticPrintConfigs
Segmentation fault (core dumped)

My hardware & configuration:
My system: Lenovo ThinkPad x230 i5, 16GB RAM, Xubuntu Linux 

LSB Version:	core-11.1.0ubuntu3-noarch:printing-11.1.0ubuntu3-noarch:security-11.1.0ubuntu3-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 21.10
Release:	21.10
Codename:	impish

server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 4.2
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL core profile version string: 4.2 (Core Profile) Mesa 21.2.2
OpenGL core profile shading language version string: 4.20
OpenGL version string: 3.0 Mesa 21.2.2
OpenGL shading language version string: 1.30
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 21.2.2
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00

xitix, 2021-12-05:
sorry, closing by mistake. Issue is present

bubnikv, 2021-12-06:
Did you compile yourself? If so, could you please run in debug mode and check the console for errors? In debug mode PrusaSlicer checks most of the OpenGL calls for error states.

bubnikv, 2021-12-06:
Do you get any errors reported into the console?

bubnikv, 2021-12-06:
Also you may want to enable OpenGL debugging for example by running
MESA_DEBUG=1 ./prusa-slicer

https://docs.mesa3d.org/debugging.html

bubnikv, 2021-12-06:
You may also try other Mesa configuration switches, however I don't have personal experience with them.
https://mesa-docs.readthedocs.io/en/latest/envvars.html

bubnikv, 2021-12-06:
You may also try
https://github.com/prusa3d/PrusaSlicer/issues/6342#issuecomment-817691136

bubnikv, 2021-12-06:
Also you may try to disable mesa threading
mesa_glthread=false ./prusa-slicer

tamasmeszaros, 2021-12-06:
I've also tried and cannot reproduce the issue with Manjaro Linux and Radeon rx 550, with open source drivers.

bubnikv, 2021-12-08:
@hejllukas also tried to reproduce on laptops with integrated Intel graphics card, but no issue.

bubnikv, 2021-12-08:
Also duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7274

xitix, 2021-12-14:
Not working, the issue is the same, error: "exited with code 11"

mesa_glthread=false ./PrusaSlicer-2.3.3+linux-x64-GTK3-202107161044_73f75f89a64872e3cf64fbdc85f2e13a.AppImage
ATTENTION: default value of option mesa_glthread overridden by environment.
[appimage-binfmt-bypass/bin] ERROR: child exited with code 11

xitix, 2022-01-31:
found it works in Ubuntu if disable libgl : 
LIBGL_ALWAYS_SOFTWARE=1  ...path/.../PrusaSlicer-2.4.0+linux-x64.AppImage

lukasmatena, 2023-02-06:
This is quite likely to be fixed in [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3). Let us know if it isn't. Closing.

closed 2023-02-06 by lukasmatena