prusa3d/PrusaSlicer #7414: Skinny textbox Beta 4

bbinnard, 2021-12-04:
### Version
2.4.0 Beta3

### Operating system type + version
Windows 11/64 Home with Intel B630 video driver (no video card) running on 4K 3840x2160 screen

### Behavior
Infill textbox is too short when un-selected. It looks OK when selected. It's been like this for quite a few releases - not a big deal but....would be nice if it got fixed. :)

[image]

bubnikv, 2021-12-08:
> It's been like this for quite a few releases - not a big deal but....would be nice if it got fixed. :)

Unfortunately we are unable to reproduce your issue, at least not at this page.

bubnikv, 2021-12-08:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/7176, which only @YuSanka could reproduce.

bubnikv, 2021-12-10:
Hopefully fixed with 63a3971091edf8510840de36e7e41ea1026eebe1

    Windows specific: Fixes layout issues on plater, short BitmapComboBoxes
    with some Windows scaling, see GH issue #7414.
    The issue was most likely due to the initial main frame size
    generated by Windows based on some unknown rule based on font scaling
    and possibly display size and something else. If the initial window height
    is short, then the last controls in a "rubbery" sizer are trimmed
    and wxBitmapComboBoxes then remember the initial widget size indefinietly.

Fun fact: I was able to reproduce the issue on my computer with 175% font scaling only and only in SLA mode.
Fun fact 2: It took me around a working day to investigate the issue.

bbinnard, 2021-12-10:
This is probably more info than you want to know , but this is a strange one so here it is: 

My system has 2 monitors. Primary is a 4K 3840x2160 landscape screen running with font scaling @ 150%. Secondary is a 1920x1200 screen in portrait mode running with font scaling @ 100%, Both of those are the Windows "recommended" scaling factors and work well across all my applications.

When I run Beta3 on my primary monitor only the Infill text box is small - everything else is A-OK. But - when I resize the Beta3 window and move it onto the secondary screen (I have to make it a lot narrower because of the portrait orientation) the Infill box appears OK. The crazy thing is if I then move the window back to the primary monitor and make it full screen, which is how I normally run, the Infill box remains full size. But - if I close and reopen Beta3 it is small again.

As I said before, my system has no video card; I run both monitors off the motherboard (Asus B360M) video connectors and my video driver is the default Intel UHD Graphics 630 one. All my software is up to date. 

Finally, ver. 2.3.3 does not have this problem on either monitor.

bubnikv, 2021-12-10:
Would you please retest with

https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta3+39-win64-g26008adfa-202112100858.zip

You may need to enter 

user prusaslicer
password slicer

if a login window opens

Thank you.

bbinnard, 2021-12-10:
That version fixes it. Way to go. I"m closing this one.

closed 2021-12-10 by bbinnard

bubnikv, 2021-12-11:
thanks a lot for testing.